mattdibi/redox-keyboard #1: Could this design be modified to allow swapable switches?

dmgm, 2018-05-08:
I have another keyboard (an Input Club K-Type) which allows you to replace switches. As a result I have a couple of sets of plate mounted rather than pcb mount switches. I'd like to try making my own keyboard, but this feature is missing from all current kit designs. I note the Ergodox EZ has this feature.

https://ergodox-ez.com/pages/change-it-yourself

mattdibi, 2018-05-08:
Hi,
it is possible. The current switch footprints must be replaced with a footprint that supports Kailh PCB sockets and is dual-sided. Save for mechanical interferences it might work.
I could consider it for a future revision.

closed 2018-05-08 by mattdibi

tstumm, 2018-08-20:
Is there any update on this? Especially if it is possible to use the Kailh sockets with the cases from Falbatech!

mattdibi, 2018-08-20:
No update. The current PCB revision doesn't support Kalih sockets.