mattdibi/redox-keyboard #2: Missing Detail in Length of Extension for 3d printed case.

irfn, 2018-07-08:
>> use the diodes legs leftover from previous steps as this will make the controller's USB port much lower, this is needed for the 3D printed case. 
This is mentioned in the readme. Can we make this more specific. I am trying to mod the case and needed this detail.

mattdibi, 2018-07-09:
Sure. What do you need in detail (measures, more detailed pics, a more detailed description...)?

irfn, 2018-07-09:
@mattdibi Need more measurements. I could find longer header pins if i know how above the controller needs to sit.

mattdibi, 2018-07-09:
I clarified the description and added some measurements. Is this enough?

irfn, 2018-07-10:
@mattdibi this is what i was looking for. You can close the issue. Thanks for your help.

mattdibi, 2018-07-10:
Great. Good luck with your project! :)

closed 2018-07-10 by mattdibi