mattdibi/redox-keyboard #3: Which version was used to build the internal redox_rev1_default_RGB.hex ?

rafaelcaricio, 2018-07-16:
Could you please let me know which release of QMK https://github.com/qmk/qmk_firmware/releases was used to build the .hex firmware present in this repository?

rafaelcaricio, 2018-07-16:
I'm trying to compile from master and burn it to a redox I've got from FalbaTech. But only the .hex file in this repository works. I've also tried to compile from multiple versions close to the commit date of the file present here, but no lucky.

mattdibi, 2018-07-17:
It shouldn't be a versioning problem. Could you be more specific about the issue? What doesn't work? The RGB lighting? Keypresses won't register? You cannot compile? To enable RGB lighting you have to follow [these steps](https://github.com/mattdibi/redox-keyboard#rgb-underglow-firmware), did you follow them? I hope this helps! :D

rafaelcaricio, 2018-07-17:
Hi @mattdibi,

I've installed all necessary dependencies to compile the QMK firmware in my MacOS.
```
make redox/rev1:default:avrdude
QMK Firmware 0.6.32
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making redox/rev1 with keymap default and target avrdude

avr-gcc (GCC) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22712	      0	  22712	   58b8	./.build/redox_rev1_default.hex

Compiling: keyboards/redox/matrix.c                                                                 [OK]
Compiling: keyboards/redox/i2c.c                                                                    [OK]
Compiling: keyboards/redox/split_util.c                                                             [OK]
Compiling: keyboards/redox/serial.c                                                                 [OK]
Compiling: keyboards/redox/redox.c                                                                  [OK]
Compiling: keyboards/redox/rev1/rev1.c                                                              [OK]
Compiling: keyboards/redox/keymaps/default/keymap.c                                                 [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_leader.c                                                 [OK]
Compiling: quantum/rgblight.c                                                                       [OK]
Compiling: ./drivers/avr/ws2812.c                                                                   [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: ./tmk_core/common/host.c                                                                 [OK]
Compiling: ./tmk_core/common/keyboard.c                                                             [OK]
Compiling: ./tmk_core/common/action.c                                                               [OK]
Compiling: ./tmk_core/common/action_tapping.c                                                       [OK]
Compiling: ./tmk_core/common/action_macro.c                                                         [OK]
Compiling: ./tmk_core/common/action_layer.c                                                         [OK]
Compiling: ./tmk_core/common/action_util.c                                                          [OK]
Compiling: ./tmk_core/common/print.c                                                                [OK]
Compiling: ./tmk_core/common/debug.c                                                                [OK]
Compiling: ./tmk_core/common/util.c                                                                 [OK]
Compiling: ./tmk_core/common/eeconfig.c                                                             [OK]
Compiling: ./tmk_core/common/report.c                                                               [OK]
Compiling: ./tmk_core/common/avr/suspend.c                                                          [OK]
Compiling: ./tmk_core/common/avr/timer.c                                                            [OK]
Compiling: ./tmk_core/common/avr/bootloader.c                                                       [OK]
Compiling: ./tmk_core/common/magic.c                                                                [OK]
Compiling: ./tmk_core/common/mousekey.c                                                             [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Compiling: ./tmk_core/protocol/lufa/lufa.c                                                          [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: ./tmk_core/protocol/lufa/outputselect.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
[... cut ...]
As you can see in the console output everything seems to compile without problems and I flash the firmware to both sides of the keyboard with no issues.

The problem is that only the right hand side works normally, RGB leds and everything. The left side, the RGB leds works fine too, but none of the keys work. When I flash the firmware present in this repo everything works as expected. That's why I am asking here which version was used to compile this `.hex`. Is there any more information you might find useful to troubleshoot this issue?

mattdibi, 2018-07-17:
Did you set in the `qmk_firmware/keyboards/redox/keymaps/default/config.h` to use serial communication?

rafaelcaricio, 2018-07-17:
Aha! No, I've not. I tried now and everything is working. Thank you very much!

closed 2018-07-17 by rafaelcaricio

mattdibi, 2018-07-17:
Great! Have fun with your Redox! =D