mattdibi/redox-keyboard #4: PCB

tstumm, 2018-08-20:
Is there any consideration to open-source the PCB shown in the README?

mattdibi, 2018-08-20:
Hi,
I need to talk about this with the producer (Falbatech). We'll see.

dmgm, 2018-08-21:
I note the project is released under an MIT license. @mattdibi has contributed the schematic, someone else contributed a design for a 3d case. It seems not to be in the spirit of the project if the PCB designer has not contributed their work under the same license.

If Falbatech is unwilling to contribute the design please consider accepting an official design from someone who will provide an open source design. 

The Ergodox is open source, but has not been updated in a while. The Infinity Ergodox is open source. The Altreus is open source. I think some people will chose this keyboard over others for that reason. I understand Falbatech is running a business, but I see no reason why people would not choose to purchase hardware from a business that supports open source, that spend more on printing their own PCB, unless transports costs and/or markup make that viable.

mattdibi, 2018-08-21:
I may have worded poorly my answer, I'm sorry.

I am the designer of the PCB, the schematic, [the original 3D printable case](https://www.thingiverse.com/thing:2886662) and the original prototype. Falbatech only manufactures and sells the PCBs and his cases.

I wanted to talk about this to him as a courtesy, since open-sourcing the PCB files might impact him directly.

tstumm, 2018-08-23:
Any news on this?

mattdibi, 2018-08-23:
I wrote to him but he didn't answer. I'll write him again tonight, but if i don't get an answer I'll proceed with the "silent consent" rule.

mattdibi, 2018-08-23:
PCB files uploaded.

closed 2018-08-23 by mattdibi