mattdibi/redox-keyboard #7: Reset button not working?

nicolai86, 2018-10-13:
Hey,

thank you for this reduced ErgoDox design, I think I'm really going to like it.

My FalbaTech order arrived today. I've soldered everything according to the build guide. However, when I try to flash the micros it seems the reset button is not working:

```
$ make redox/rev1:default:avrdude
QMK Firmware 0.6.140
Making redox/rev1 with keymap default and target avrdude

avr-gcc (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  23468	      0	  23468	   5bac	.build/redox_rev1_default.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/redox_rev1_default.elf                                                              [OK]
Creating load file for flashing: .build/redox_rev1_default.hex                                      [OK]
Checking file size of redox_rev1_default.hex                                                        [OK]
 * The firmware size is fine - 23468/28672 (5204 bytes free)
Copying redox_rev1_default.hex to qmk_firmware folder                                               [OK]
Detecting USB port, reset your controller now..................................................................................................................................................................................................................................................................................................................................^Cmake[1]: *** [avrdude] Interrupt: 2
make: *** [redox/rev1:default:avrdude] Interrupt: 2
```

I tried various things to get the reset buttons to work - in the end I just shorted `GND` and `RESET`. Do you have any ideas how to address the issue with the reset buttons?

Best,

nicolai86, 2018-10-14:
I think I've found the culprit, closing for now.

mattdibi, 2018-10-14:
Hi,
I'm glad you resolved the problem. May I ask you what was the issue and how you solved it? It may help others with the same problem.

nicolai86, 2018-10-14:
I did not properly solder some of the micro headers on the front of the PCBs. I started checking the SMD diodes using a multimeter while the keyboard was attached to my laptop and they were missing voltage. I’m sorry to say that the root cause was my sloppy soldering; 
I was able to fix this by desoldering every single switch as well as the pro micros, bc there was no other way to access these solder points with the case attached.

If you have suggestions on how to eg replace the switches below the pro micro or the pro micro itself; would be a very useful addition to the guide for “repairs” like mine

nicolai86, 2018-10-14:
@mattdibi I'm re-opening this as it seems I'm missing more:

after correcting the soldering points the reset buttons work fine, flashing works fine - when I connect each half to my laptop, both halves work fine. However, when I connect them via TRRS, the half not connected via USB does not work. As I've got a RGB underglow mod I've tried removing it and reverting to I2C instead of serial, but no success.

I verified with a multimeter that the TRRS line is carrying ~5V. I expect the other micro to light its green light, but this only happens when connected via USB. Any ideas what I am missing?

nicolai86, 2018-10-14:
I've flashed the EEPROM on both hands; now every hand has the right keys when used alone, but when attached via TRRS still only one side works.

mattdibi, 2018-10-14:
You are correct in assuming that the Arduino should light up when connected through the TRRS connection. I suggest you to double check the connection of the TRRS jack and the cable. Every pole of the jack should correspond to the same pole on the othe end of the cable.
Also check for continuity to the arduino: the top pin of the TRRS connector should connect to ground, the bottom one to VCC.
A picture of the PCB might help.

nicolai86, 2018-10-16:
@mattdibi so I've replaced the pro micro now, here are some pictures:

left hand:
[image]

right hand:
[image]

both hands work individually without any issues.

nicolai86, 2018-10-16:
Now, for the TRRS debugging: right hand has a fine tip and bottom I think:

[image]

However, left hand seems to have VDD on the round, not the tip:

[image]

Could the PJ320 be defect?

nicolai86, 2018-10-16:
desoldered the PJ320A, I think it works fine. I think there's an issue with the pins actually:
[image]

According to my multimeter it's

T R (VDD) R S (GND), but the right hand side is T(VDD) R R S (GND)

mattdibi, 2018-10-16:
You still need to define `EE_HANDS` in `keyboard/redox/keymaps/<your_keymap_of_choice>/config.h` like so:

```c
/* Select hand configuration */

// #define MASTER_LEFT
// #define MASTER_RIGHT
#define EE_HANDS

#undef RGBLED_NUM
```

Full instruction are [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox#setting-ee_hands-to-use-either-hands-as-master).

Also double check the connectivity between PD0 and the TRRS pin below the one you soldered the cable for the RGB lighting.

nicolai86, 2018-10-17:
I've checked continuity for PD0 and PD1, both are correct on both the PJ320A and the inserted TRRS cable - I'll flash the micros and let you know if that helps!

mattdibi, 2018-10-17:
I think this is a software issue too. Try this: in this repository for the Redox rev.1 I uploaded a pre-built hex file for RGB enabled version [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox/firmware). Try using that hex and see if it works.

nicolai86, 2018-10-17:
@mattdibi hm, I'm confused. I've flashed each half individually with the RGB hex file you linked:

```
$ avrdude -p atmega32u4 -P /dev/tty.usbmodem14501 -c avr109 -U flash:w:redox_rev1_default_RGB.hex
```

both times flashing  worked without any issues.

Now I connect the keyboards via TRRS, attach one half via USB - and still, only one half works. 
So, maybe a hardware issue after all? Is there any way I could debug the 1-wire connection between each half?

mattdibi, 2018-10-17:
OK this is getting weird. Have you tried to disable the RGB lighting? I'm trying to think a way to debug this. When I come home from work tonight I'll try to reproduce the problem on my debugging board.

nicolai86, 2018-10-17:
I’ll solder the 4.7k resistors on the left side tomorrow (11:15pm here) and flash both halves with the default hex from the folder you linked.

Will report back with the results.

lleonini, 2018-10-18:
Hi,
I have the same problem.
I have ordered a built keyboard at falbatech.
Then after 2 days, one microusb plug was defective and detach. So I have just replace the pro micro by a new one. Now, as you, I can flash each hand separately and connect my keyboard via 2 micro usb and basically it works. But connected with TRRS (tested with microusb on each side) it will not work anymore (the part connected via TRRS is not seen). However when I plug it I have a the leds lines and pro micro led on on of both side.
I don't really understand what have changed.
Thanks

nicolai86, 2018-10-18:
So I’ve soldered the resistors flashed the default hex without RGB, same result. Currently disassembling the entire keyboard to find a flaw; so far, no luck. 
Hope to know more later this weekend.

lleonini, 2018-10-19:
My side I have fixed the problem with these steps:

1) On right part:
sudo avrdude -p atmega32u4 -P /dev/ttyACM0 -c avr109 -U flash:w:redox_rev1_default_RGB.hex -U eeprom:w:keyboards/lets_split/eeprom-righthand.eep

2) On left part:
sudo avrdude -p atmega32u4 -P /dev/ttyACM0 -c avr109 -U flash:w:redox_rev1_default_RGB.hex -U eeprom:w:keyboards/lets_split/eeprom-leftthand.eep

Then it works.

3) Then to reflash my custom config I have used:
EE_HANDS + USE_SERIAL in config.h

Hope this help you, in my case that was definitely a software issue.

mattdibi, 2018-10-19:
@lleonini Thanks for sharing. I'm glad you solved your problem.

@nicolai86 I still have no clue, neither was able to reproduce the problem. Please keep me updated.

mattdibi, 2018-10-19:
@nicolai86 I'm sorry to hear that. If you need support for your next kit or some instruction is not clear feel free to ask.

nicolai86, 2018-10-19:
Will do, thank you @mattdibi :)

nicolai86, 2018-11-09:
@mattdibi for what it’s worth I figured out the issue: the TRRS cable from FalbaTech was broken; unfortunately I have already thrown away the soldered PCB, but my second Redox works really well . Hah!

mattdibi, 2018-11-09:
@nicolai86 Well at least you found the culprit. Thank you for letting me know!

luistorres, 2019-06-25:
Hi there. 
I just ordered a fully assembled redox keyboard from falbatech. I guess it comes with the default keymaps, but some keys are not really what they should be, so I'm trying to flash it with the default image. My problem is I can't make the keyboard enter the `RESET` mode...somehow the key combinations are not working. Is there any other way to do it? Preferably one that doesn't involve disassembling it?
Thanks 🙏

mattdibi, 2019-06-26:
Hi,
on the back of the keyboard there's a button for this purpose. Hope this helps.

P s. Next time or if this doesn't help you, please open a new issue.