mattdibi/redox-keyboard #8: Middle row (a,s,d,f,...) causes whole column to trigger

tstumm, 2018-10-15:
I just finished soldering my Redox. After flashing I noticed that pressing a middle row key causes a press on all switches in the same column. 

The other half does not have the same issue. I'm quite sure that the diodes are correctly soldered, but I don't have a clue what could be the culprit here. Especially since the whole row is affected.

tstumm, 2018-10-15:
I checked the PCB using a multimeter and it seems that all rows go low upon pressing a button in the middle row.

mattdibi, 2018-10-15:
Check the soldering for a short, this shouldn't happen. Also double check the diodes orientation.

mattdibi, 2018-10-15:
> there are only two connected pins, which is intended.

Could you elaborate on this?

tstumm, 2018-10-16:
I meant the GND pins, but I found another pair of pins that were shorted. I guess the Pro Micro didn't survive getting unsoldered when I soldered it in the wrong way around the first time.

Replacing the Pro Micro solved the issue :)

closed 2018-10-16 by tstumm

mattdibi, 2018-10-16:
I'm glad you solved the issue. Enjoy your Redox! :smile: