mattdibi/redox-keyboard #10: RGB Underglow problems

manno, 2018-10-29:
I'm trying to get RGB underglow to work, but when I follow the instructions it only works on master.
If I recompile and re-wire the strip connections can switch master to the other hand, though.

If I had to guess, I'd say it's either the firmware or the TRRS connection?

mattdibi, 2018-10-29:
I'm sorry: by 

>  it only works on master.

do you mean that only master hand lights up?

manno, 2018-10-29:
Yes, I can control animations etc on master.

mattdibi, 2018-10-29:
The keys on the slave half are working?

manno, 2018-10-29:
Oh yes, the keys are working.

mattdibi, 2018-10-29:
Did you set the communication to serial as per [this](https://github.com/mattdibi/redox-keyboard/tree/master/redox#rgb-underglow-firmware) instructions?

manno, 2018-10-29:
Yes, https://gist.github.com/manno/07faacd854cf74a3468fa45f63fffb7b

manno, 2018-10-29:
Haha, I thought there were three 😢 
Thanks, will try to get some solder in..

manno, 2018-10-29:
Managed to solder the missing pin. However on one jack the two ring pads seem to be connected. I already soldered the switches and can't reach those. Maybe I fix that once I have access to a desoldering gun...

Thanks for the help and all the work going into Redox 👍

closed 2018-10-29 by manno

mattdibi, 2018-10-29:
I'm glad I was able to help!