mattdibi/redox-keyboard #11: Integration of trackpoint - possible?

hgersen, 2018-12-31:
The design of the redox keyboard without any cables looks really, really nice and I would love to replicate that as the cables have always been a source of problems and tend to get in the way. However I had promised myself that my next keyboard would integrate a trackpointer as that is one of the key things I really miss having used IBM/Lenovo laptops for a long time.

Incorporating a trackpointer is supported by the QMK firmware and is relatively well documented for directly connecting a salvaged trackpointer to the teensy board. Background info:

QMK_docs: https://github.com/qmk/qmk_firmware/blob/master/docs/feature_ps2_mouse.md
walkthrough with pinouts and the like: https://github.com/alonswartz/trackpoint#tmk-firmware-changes-and-tweaks
example on ergodox: https://geekhack.org/index.php?action=dlattach;topic=82599.0;attach=138751;image (I would leave out the mouse-buttons & put the trackpointer between fgv keys instead)

Looking through the code for the redox-firmware I struggle to see whether integrating a trackpointer with the wireless version would be feasible although the Core51822B in the transmitter seems to have the UART still free; yet I guess that would still need (a lot of?) dedicated code to get the PS/2 key codes to the Arduino board. Something that is probably beyond me. 

As I personally would find it a great enhancement/feature. Do you have any insights in whether integrating a trackpointer in the wireless version is even feasible? If yes: how hard would that be?

mattdibi, 2018-12-31:
Pretty hard. You'll need to:
- Modify the PCB
- Modify the case
- Add a booster circuit (the transmitters are 3V,  the trackpoint seems to work on 5V)
- Add the code to handle the trackpoint
- Modify the data sent to the receiver to handle the trackpoint
- Modify the code of the receiver to send this data to the Arduino
- Adapt the Arduino Firmware accordingly

This without considering that you cannot directly use the QMK support for the trackpoint because of the technique used to send the keyboard states over the air. The receiver communicates the keyboard state to the arduino through UART, so you need to find a way to send the trackpoint data through the air, and to communicate it to the Arduino.

hgersen, 2019-01-01:
Thanks for looking into this and your answer is as I feared. The amount of new code needed would make this a major project seeing I have little experience with programming Arduinos. I'll stick with the wired option for the next build for now. :(

closed 2019-01-01 by mattdibi