mattdibi/redox-keyboard #13: RGB Underglow not working

Sanic, 2019-01-03:
Hi!

First things first: Thank you very much for this nice project. I built a wired redox over the couple last weeks and i'm close to finishing this project. The keyboard functionality itself does work on both halves, but i have problems getting the RGB underflow (6 LEDs per half) working.
I've flashed both halves with the same firmware using the following config.h:

````
/* Use I2C or Serial, not both */
#define USE_SERIAL
// #define USE_I2C

/* Select hand configuration */
#define MASTER_LEFT
//#define MASTER_RIGHT
// #define EE_HANDS

#undef RGBLED_NUM
#define RGBLIGHT_ANIMATIONS
#define RGBLED_NUM 12
#define RGBLIGHT_HUE_STEP 8
#define RGBLIGHT_SAT_STEP 8
#define RGBLIGHT_VAL_STEP 8
````
I didn't write the EEPROM content. This isn't needed when i'm not using EE_HANDS, right?

Right after flashing, the RGB underflow didn't work. Should it light up per default?
After checking the current keymap in the qmk repo, i saw that the light can be toggled on the ADJUST layer with RGB_TOG. I've tried that but it didn't help. Other keys (such as F3, F4) are sent fine, so i think that i'm on the right layer.
The rules.mk is still on the default values:
````
BOOTMAGIC_ENABLE = no      # Virtual DIP switch configuration(+1000)
MOUSEKEY_ENABLE = yes       # Mouse keys(+4700)
EXTRAKEY_ENABLE = yes       # Audio control and System control(+450)
CONSOLE_ENABLE = no         # Console for debug(+400)
COMMAND_ENABLE = yes        # Commands for debug and configuration
# Do not enable SLEEP_LED_ENABLE. it uses the same timer as BACKLIGHT_ENABLE
SLEEP_LED_ENABLE = no       # Breathing sleep LED during USB suspend
# if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ#nkro-doesnt-work
NKRO_ENABLE = no            # USB Nkey Rollover
BACKLIGHT_ENABLE = no       # Enable keyboard backlight functionality on B7 by default
MIDI_ENABLE = no            # MIDI support (+2400 to 4200, depending on config)
UNICODE_ENABLE = no         # Unicode
BLUETOOTH_ENABLE = no       # Enable Bluetooth with the Adafruit EZ-Key HID
AUDIO_ENABLE = no           # Audio output on port C6
FAUXCLICKY_ENABLE = no      # Use buzzer to emulate clicky switches
RGBLIGHT_ENABLE = yes       # Enable WS2812 RGB underlight. 

SPLIT_KEYBOARD = yes

DEFAULT_FOLDER = redox/rev1
````

Hardware-wise, i checked that the RGB strips have +5 Power and i also checked that TX0 on the left hand master is connected to the DIN of the RGB strip in the left hand.

Do you have any suggestions how i can debug this further?

mattdibi, 2019-01-03:
Hi,
you seem to have everything set correctly.

> I didn't write the EEPROM content. This isn't needed when i'm not using EE_HANDS, right?

You're correct, it isn't needed.

I have two suggestions:
1. As of the latest pull request the RGB backlighing is supported by default. Compiling and unloading the default keymap without any modifications should lights the LEDs up.
2. [Here](https://github.com/mattdibi/redox-keyboard/tree/master/redox/firmware) you can find a precompiled  hex with rgb support to double check if it is a software problem.

I neither of the above work it should be a hardware problem.

mattdibi, 2019-01-04:
Great! Glad I was able to help :smile: