mattdibi/redox-keyboard #15: Add Mill-Max and Peel-A-Way hotswap details to rev 1 README.md

unjordy, 2019-02-09:
I've successfully assembled a fully hot-swappable Redox with Falba's PCBs and cases, Mill-Max 7305 sockets for the switches, and Peel-A-Way sockets for each microcontroller. The process is fairly simple, but it involves a couple of changes to how one would normally assemble the keyboard. I've taken several photos detailing how the sockets should fit. If you're interested adding the steps to the Redox rev 1 README, would you prefer a pull request, or a separate document you could edit into the existing README?

mattdibi, 2019-02-09:
That's awesome! I would be really happy to add your instructions to the repo.

I think a pull request would be great, this way you will be credited for your work. :D

You can add your instructions in the rev1 README adding a new section under the name of "Hot swappable mod". At the start of the section add the bill of materials for the mod with a link to a store. Put all the images in the `img/` folder with a recognizable prefix (like `hs-mod-blablabla.png`).

I think that's all. :)

unjordy, 2019-02-18:
All set! Check out PR #17 and let me know how it looks. And thank you by the way for the excellent keyboard design! This was my first build, and it's been excellent so far. Next up, I want to figure out how to adhere the RGB underglow strips to the PCB in such a way that they won't interfere with the hot swap sockets.

mattdibi, 2019-02-18:
Merged.

Thank you for your contribution. It looks great! :)

unjordy, 2019-02-22:
Glad to help! Writing this commit was an excellent excuse to use my Redox and tune in its layout. I'm always a bit surprised how quick I can already type on it.

closed 2019-02-22 by unjordy

ArunJ98, 2021-03-15:
@unjordy Hey! I was just wondering if you were able to figure out a way to make the RGB underglow work with the hot swappable setup? Thanks!

unjordy, 2021-03-15:
@ArunJ98 Hey! I haven’t had the chance to try yet — my Redox became my daily driver, so there’s been no opportunity to open it up since I assembled it. Zero downtime is definitely good news for the longevity of Falba’s parts and the hot swap mod though!