mattdibi/redox-keyboard #19: EE_HANDS eep files

fculpo, 2019-03-20:
Hi,

The specified .eep files for flashing eeprom are not anymore in QMK codebase: 
`avrdude -p atmega32u4 -P $(COM_PORT) -c avr109 -U eeprom:w:keyboards/lets_split/eeprom-lefthand.eep`
```

avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
avrdude.exe: reading input file "keyboards/lets_split/eeprom-lefthand.eep"
avrdude.exe: error opening keyboards/lets_split/eeprom-lefthand.eep: No such file or directory
avrdude.exe: can't determine file format for keyboards/lets_split/eeprom-lefthand.eep, specify explicitly
avrdude.exe: read from file 'keyboards/lets_split/eeprom-lefthand.eep' failed
```

Some are available in `/path/to/qmk/quantum/split_common/`.
Can we use them ? I fyes, could you update the documentation ?

mattdibi, 2019-03-20:
Hi,
thanks for the info. I'll investigate this during the weekend and update the documentation accordingly. :)

mattdibi, 2019-03-23:
Fixed! ;)

closed 2019-03-23 by mattdibi