mattdibi/redox-keyboard #22: redox-w question: Have you had issues with some columns not registering when key from nearby column is held?

ilukinov, 2019-04-21:
Hello, thanks for your project, it helped me a lot with my own custom keyboard.
I'm using wiring similar to yours (you don't use pin 8), 
[image]
but for some reason I'm having weird issue with neighboring columns.  When I holding down key from one column another one(neighboring) won't work. 
For example:
- if I holding shift(c0r3) key then none of keys from c1 will work (1 q a z).
- if I holding z(c1r3) key then none of keys from c2 work (2 w s x) but c0 keys will work just fine as well as all other keys, so keys from c0, c1, c3, c4, c5, c6 will still work.

Have you had any issues like that? Can you test such scenarios on your keyboard (left part)

Thank you.

mattdibi, 2019-04-22:
Hi,
I've never came encountered such behavior. Double check your soldering, it looks a hardware problem to me.

ilukinov, 2019-04-22:
The weird part is that when I'm hooking up same matrix to promicro all works just fine. That's what get's me confused.

mattdibi, 2019-04-22:
> when I'm hooking up same matrix to promicro all works just fine

...that's weird. Are absolutely sure everything stays the same between switching from the pro micro to the core5122? If it works for the Pro Micro it should work for the Core51822 too...

ilukinov, 2019-06-14:
Well, since no one has similar issues I'm closing it. Maybe it's just my chip revision or something.

closed 2019-06-14 by ilukinov