mattdibi/redox-keyboard #24: 4th LED on Redox-W receiver ?

LucidityCrash, 2019-05-11:
Please tell me what it's for, it's driving me crazy ... I can't see where it is used in QMK ?????

mattdibi, 2019-05-11:
Hi,
You're right I'm sorry. I should have stated it somewhere... currently it's not used by the default QMK firmware, but you can easily make it work by modifying the sources.

LucidityCrash, 2019-05-11:
I'm not really up to speed with full on Arduino coding and bit shifting so sorry if this is stupid but would I achieve this by doing :
	DDRD  |= (1<<1) | (1<<0);
	PORTD |= (1<<1) | (1<<0);

mattdibi, 2019-05-12:
I had an example laying around. Tell me if this works for you.

```diff
diff --git a/keyboards/redox_w/redox_w.c b/keyboards/redox_w/redox_w.c
index 75df910..860cfe1 100644
--- a/keyboards/redox_w/redox_w.c
+++ b/keyboards/redox_w/redox_w.c
@@ -9,6 +9,10 @@ void led_init(void) {
        PORTD |= (1<<1);
        DDRF  |= (1<<4) | (1<<5);
        PORTF |= (1<<4) | (1<<5);
+
+    // Caps lock and num lock
+    DDRB |= (1 << 4) | (1 << 5);
+    PORTB &= ~(1 << 4) | ~(1 << 5);
 }
 
 
@@ -27,7 +31,16 @@ void matrix_scan_kb(void) {
 }
 
 void led_set_kb(uint8_t usb_led) {
-
+    if (usb_led & (1<<USB_LED_NUM_LOCK)) {
+        PORTB |= (1<<4);
+    } else {
+        PORTB &= ~(1<<4);
+    }
+    if (usb_led & (1<<USB_LED_CAPS_LOCK)) {
+        PORTB |= (1<<5);
+    } else {
+        PORTB &= ~(1<<5);
+    }
 }
 
 #ifdef ONEHAND_ENABLE

```

mattdibi, 2019-05-12:
I had this piece of code laying around that should do what you're looking for. Tell me if this works for you.

```diff
diff --git a/keyboards/redox_w/redox_w.c b/keyboards/redox_w/redox_w.c
index 75df910..860cfe1 100644
--- a/keyboards/redox_w/redox_w.c
+++ b/keyboards/redox_w/redox_w.c
@@ -9,6 +9,10 @@ void led_init(void) {
        PORTD |= (1<<1);
        DDRF  |= (1<<4) | (1<<5);
        PORTF |= (1<<4) | (1<<5);
+
+    // Caps lock and num lock
+    DDRB |= (1 << 4) | (1 << 5);
+    PORTB &= ~(1 << 4) | ~(1 << 5);
 }
 
 
@@ -27,7 +31,16 @@ void matrix_scan_kb(void) {
 }
 
 void led_set_kb(uint8_t usb_led) {
-
+    if (usb_led & (1<<USB_LED_NUM_LOCK)) {
+        PORTB |= (1<<4);
+    } else {
+        PORTB &= ~(1<<4);
+    }
+    if (usb_led & (1<<USB_LED_CAPS_LOCK)) {
+        PORTB |= (1<<5);
+    } else {
+        PORTB &= ~(1<<5);
+    }
 }
 
 #ifdef ONEHAND_ENABLE

```

mattdibi, 2019-06-07:
Closing as resolved.

mattdibi, 2019-06-08:
I need to make a correction!

The correct code to use the 4th LED is [here](https://github.com/mattdibi/qmk_firmware/commit/e0d24216bcaaed86a148730b6cb74b97d3685f1d).

closed 2019-06-08 by mattdibi

pixelbreaker, 2019-08-27:
@mattdibi Thanks for the link to the other issue, I've cleaned up my code a LOT now with thenew higher level API. 👍