mattdibi/redox-keyboard #25: Slave Not Working with RGB Underglow

JustSpiffy, 2019-05-20:
Hey -

This is probably my fault. Still debugging.

I can't get the slave side to work with the new set up. I have set the config as described in the documents, using SERIAL and the proper number of LEDs. I have set it to MASTER_RIGHT. I have not flashed the EEPROM yet.

It seemed that flashing the EEPROM was unnecessary if you don't set EE_HANDS.

I flashed both sides with the same firmware. Connecting the cable to both sides, shows both sides working (though using right hand keys since I flashed right master on both). 

I'm going to check continuity along the two TRRS connectors but I was wondering if there was a recent bug fix in the firmware or something?

P.S. - Right hand should be "KC_SLASH" based on your diagram at the bottom instead of "KC_BKSL"

mattdibi, 2019-05-21:
Hi,
I would suggest you to double check the continuity on the two TRRS connector. Software side you seem to be ok, and there shouldn't be any recent bug fix. Be warned about [this](http://www.40percent.club/2017/09/green-pro-micro.html?m=1)

JustSpiffy, 2019-05-21:
Mattia,

I actually fixed this yesterday, it was TRRS continuity. The falbatech case prevented the cable from going in all the way. A little bit of dremel magic fixed the situation.

Thanks very much for the support!

closed 2019-05-21 by JustSpiffy

mattdibi, 2019-05-21:
Nice! I'm glad you fixed it. Enjoy your keyboard :)