mattdibi/redox-keyboard #27: What happens if I connect each half, with interconnect, to the computer?

minjaesong, 2019-05-27:
So basically:
```
 [ USB PORTS ]
  ||       ||
[LEFT]===[RIGHT]
```
The keyboard would work fine (with some delay) *without the interconnect*, but I'm afraid to try it with the interconnect. Would the keyboard still work? would it fry something?

mattdibi, 2019-05-27:
Before answering I would like to ask you a question: what would be the purpose of such a configuration? There may be other ways to achive what you want to achive...

minjaesong, 2019-05-27:
Just a silly question I had for a while, sorry if my question was disturbing. Obviously I didn't tried it as I'm thinking it may outright break the keyboards; then again I'm not very familiar with the electronics.

mattdibi, 2019-05-27:
Okay then.

In theory, having the power the same reference, both the Arduinos should see the same power. They would just power up normally, problems may arise from software side since both Arduinos will think they're the master (seeing that they are powered through USB).

I personally didn't try and advice you against it, but I don't think anything catastrophic should happen.

minjaesong, 2019-05-28:
Okay then. Thank you for the answer!

closed 2019-05-28 by minjaesong