mattdibi/redox-keyboard #30: RGB underglow on redox_w

Pastitas, 2019-06-04:
Hi! currently writing this from my newly built Redox rev-1 (with wich i'm falling in love, great design!) i was wondering if you have thought about a redox wireless with rgb underglow. I have found this [repo](https://github.com/joric/jorian) that achieves somewhat the same thing. 

So, have you tought about it? maybe i could get it working.
Also haven't really looked into mitosis firmware deep, would a hardware mod be necesary for this to work? it does not look like it would from what i've seen in that repo

 Thanks in advance, best regards!.

mattdibi, 2019-06-04:
Hi,

I considered it and discarded the idea as the power consumption would skyrocket. Right now we have an average current draw of [around 250 uA](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w#battery-usage). Adding an RGB strip WS2812 based would mean adding 40 mA PER LED! Remember you only have 2200 mAh of battery! 5 LEDs would kill the battery in 10h (in reality this number is lower because the MCUs works for 1.7-3.3V range).

If you want the LEDs that bad you actually need some hardware tinkering since there's the need to carry power (remember that the transmitters are 3.3V whereas WS2812 typically are 5V based) and data from the MCUs to drive them.

Hope this answer your question.

mattdibi, 2019-06-07:
Closing as resolved.

closed 2019-06-07 by mattdibi