mattdibi/redox-keyboard #31: Wireless range issues

InvaderSquibs, 2019-06-06:
I have been having issues with only one side of my redox wireless. The right side can go well over a few feet with no missed keys, but the left side has to be within a foot for the receiver to pick it up. I know it may be a shot in the dark but any ideas of what I can try to trouble shoot? (I have already tried mucking about with the batteries and know that isn't the problem)

mattdibi, 2019-06-06:
Maybe related to [this](https://github.com/mattdibi/redox-w-firmware/issues/3#issue-442735897). Still haven't found the issue.

Are you using the aluminium case?

InvaderSquibs, 2019-06-26:
I am using a wooden case from falbatech. I ran a few tests with it and have been finding that though the range can go pretty far but struggles with any sort of interfere  between direct line of sight.so even if I waved my hand in front of the sensor it would cut off if the sensor was over a foot away. I rearranged my setup a bit and now it rarely happens

closed 2019-06-26 by InvaderSquibs

mattdibi, 2019-06-26:
Thanks for the insight. I'm glad you found a solution.