mattdibi/redox-keyboard #32: flashed firmware... now what?

crafttherapy, 2019-06-07:
So i flashed the firmware to the receiver and both left and right sides. i connect the receiver and turn on the keyboards but nothing works. 

is there a specific sequence i need to follow? are the left and right sides automatically paired to the receiver?

mattdibi, 2019-06-07:
Did you programmed the Arduino Pro Micro?

crafttherapy, 2019-06-10:
i did do all that but the keyboard doesn't work. pretty sure everything is soldered correctly and programmed as per the build instructions. did i need to program the Arduino Pro Micro separately from the receiver section?

mattdibi, 2019-06-11:
Nope. The way the Arduino is programmed doesn't matter.

Did you turn on the two halves of the keyboard? Did you programmed the receiver?

crafttherapy, 2019-06-11:
got it working... totally didn't flash qmk firmware on the pro micro. everything works now. Thank you for amazing project.

closed 2019-06-11 by crafttherapy

mattdibi, 2019-06-11:
Glad you fixed it! Enjoy your keyboard :)