mattdibi/redox-keyboard #33: How to check TRSSS continuity ?

toindev, 2019-07-16:
Hello,

First of all, thanks for this project, even if my keyboard isn't fully functional yet, it is still looking good (Falba bamboo) :)

Now, my issue is very similar to #25, but I can't seem to find where the problem is from:
- I've flashed the QMK redox default (use serial, master left) on both sides, using QMK Toolbox
- Both sides work when plugged (although with left hand mapping)
- The leds on both sides light up in purple when everything is plugged, but the slave half keys don't work.
- I've tried several TRRS cables, and checked the cables themselves (they're OK)
- I've not soldered the resistors, as instructed in your build guide

I'm OK with code, but only so-so with electronics. Could you maybe make some kind of checklist on what to check to find what's wrong ? I'd be happy to contribute to the troubleshooting guide :D

mattdibi, 2019-07-16:
Hi,

thank you :D 

If you have a multimeter around you should check the continuity between the two halves while the TRRS cable is plugged in. There are 4 contacts inside the TRRS connector, check that the four pads on the left side connects to the 4 pads to the right side.

If everything is in order you can check if there's continuity between the two pro micros D0 pins. Then if they receive power.

toindev, 2019-07-23:
Hi,

Here is the followup: 

- TRRS continuity seems OK (I used the ground pin of the resistors next to the jack to check for ground continuity)
- Power is OK everywhere (when fully connected, controller and led strips light up)
- **D0 continuity does not seem OK**

Just to make sure, I checked for the top left pin of the controller on the right hand backside, et top right pin on the left hand backside.

So, some more questions:

- Is there a way to check for continuity from D0 to the board, to identify which side has a problem?
- I have just noticed that on your build picture of the right backside, not all ground pins are soldered. Could that be a problem?
- What should I check next? 😄

mattdibi, 2019-07-23:
Hi,
ok if D0 is not connected the two halves cannot talk to each other. This is the issue you're experiencing.

> Is there a way to check for continuity from D0 to the board, to identify which side has a problem?

The D0 is connected directly to the TRRS connector. Check the pads, at least one of the TRRS connector pad should be connected to D0 (I can't remember which one of them).

> I have just noticed that on your build picture of the right backside, not all ground pins are soldered. Could that be a problem?

What pins are you talking about? Could you provide me with a link to the picture you're referring to?

> What should I check next?

See above :smile:

toindev, 2019-07-24:
> > I have just noticed that on your build picture of the right backside, not all ground pins are soldered. Could that be a problem?
> 
> What pins are you talking about? Could you provide me with a link to the picture you're referring to?

[The two ground pins on the left side of the controller](https://github.com/mattdibi/redox-keyboard/blob/master/img/rgb-underglow2.jpg)

mattdibi, 2019-07-24:
Ah yes. Don't  worry about them. They're all connected togheter to the other GND pin. I used only one GND pin in that build because I desoldered that arduino from another project and couldn't remove the solder from that pins.

In your build you should connect all the pins of the arduino to the PCB.

toindev, 2019-08-02:
Hi,

I have spent some time properly testing, using the schematics, and am still puzzled at why the right hand can't send any key press:

* TRRS continuity is OK for all pins
* VCC to VCC has continuity
* SDA to SDA has continuity
* SCL to SCL has continuity
* Underglow, and underglow keys (from left hand) work
* Both half work when plugged directly

**The weirdest thing is that we are three people with a group order, and we all encounter the same problem.** I have tried mix and matching the halves and still could not manage to have any right half working over TRRS.

I'm not sure what I should try next, here are some of my ideas, if you have some suggestions, that would be really nice:
* A software problem: unlikely, since I used the version in this repository for tests, flashing the same on both sides.
* Remove the leds, solder the resistors and try ICC comm
* Change the soldering of the led data, and switch the master hand
* I've used pro micros ordered from an other vendor, they don't seem to have any problem, but maybe someone is aware of something that could explain our problem

Thanks in advance for any input you could have :)

mattdibi, 2019-08-04:
Hi,

This is really strange. Regarding the Pro Micro I'll copy-paste the warning I have in the BOM for the Redox.

>  :warning: Pay attention to the Pro Micro you use in this project. Online you can find the infamous Green Pro Micros which have the same pinout of the normal Pro Micro but use a different power circuitry. Being the Redox a split keyboard that relies on detecting which side is directly powered to find which side is the master, these clones are not suited for the task. A more detailed explaination [here](http://www.40percent.club/2017/09/green-pro-micro.html). Please note that they can come in different colors, not necessarily green.

Did you check this?

Another thing, are you using serial communication right? (If you're using the pre-compiled hex files you are.) Maybe double check that too. [Here](https://github.com/mattdibi/redox-keyboard/tree/master/redox#rgb-underglow-firmware)'s the firmware instructions.

toindev, 2019-08-08:
Hi,

Well, my pro-micros are blue, and seemed honest, but they may not be. I have tested all of them and all seem to have continuity between VCC and USB VCC, just has if their J1 was bridged. I have checked that this was the case using the TRRS and USB-A part of the cables plugged in every half I had on hand (5 of them).

I think this is **NOT** what we want.

So two questions for you:

* Am I right when I understand VCC and USB VCC should not be bridged?
* Is is possible to flash a custom firmware on each half as a workaround to force recognition of the slave half, or would we run into other problems? (Please say yes, we are not afraid of software, but it would really suck to have to unsolder 6 controllers 😢 )

mattdibi, 2019-08-08:
Hi,

could you send me a pic of those controllers?

Anyway during the weekend I'll try to come up with a solution. ;)

mattdibi, 2019-08-10:
Hi,

- You're right. VCC and USB VCC should not be shorted.
- A custom firmware will do just fine,  you'll need to change the `has_usb()` function to return always true for the plugged in half and always false for the other.

toindev, 2019-08-15:
Great news! I was starting to turn crazy with those :smile:

I'll send you a pic of the controller and will try a custom firmware next week and let you know how it goes.

Thanks a lot!

toindev, 2019-08-21:
Did not think about simply sending you the link, which has the best pictures you'll get:

https://www.amazon.fr/gp/product/B07FQBQ4Z6/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

toindev, 2019-08-21:
Regarding the hardcoded firmware, I've had no luck yet:

Tried to flash with the following modifications:

```
bool is_keyboard_master(void)
{
  return true;
}
```

```
bool is_keyboard_master(void)
{
  return false;
}
```
 The master works, but not the slave, just the same.
The slave half works as a left hand when flashed with the master firware as well.

I will try with the other keyboard to make sure, and then maybe have a look at how to live debug (this could be interesting).

If you have other ideas, I'm open, else I'll follow up here.

mattdibi, 2019-08-22:
@toindev This is getting weird.

I took a look at the link you sent me: the Arduinos looks good, the power circuitry is correct. This means a software fix shouldn't be needed. Maybe J1 is shorted? (it shouldn't for the split SW to work)

Regarding the software you modified: could you list  the commands you used to compile and upload the firmware to the keyboard?

For live debugging take a look [here](https://beta.docs.qmk.fm/newbs/newbs_testing_debugging).

toindev, 2019-08-22:
https://github.com/toindev/qmk_firmware/tree/hardcoded_master_slave

There's only the last commit with my modifications:

- copied the default keymap as a slave keymap
- modified the dockerfile to build both keymaps
- modified split_common to return true or false depending on a macro being set in config.h

Commands for build (powershell syntax for the Docker volume):

```
docker build --rm -f "Dockerfile" -t redox-builder .
mkdir output
cd output
docker run -i --rm -v ${PWD}:/artifacts redox-builder
```

yoution, 2019-12-14:
Hi, I have the same problem If I use two usb cables, they work well, but If I use one trss cable and one usb cable, the slave half not work, I want to test the D0, but how to test D0?can you tell me how to test D0?

mattdibi, 2019-12-14:
Hi @yoution,

First thing first, I have to tell you that the soldering job on the Arduino Pro Micro is really poor. Don't worry, it's easy to fix. I suggest you to take a look at [this](https://www.youtube.com/watch?v=fYz5nIHH0iY) tutorial and just do a new pass on all those pins (you don't need flux, just look at his final result for reference). That can be enough to fix the problem.

Another check I suggest you to do is the continuity on the cable. Put the multimeter in continuity mode and check that every ring of the 1.5mm jack is connected with the other 1.5mm jack.

Then check the continuity on Pro Micros. While the TRRS cable is connected check that PD0 on the master side is connected to PD0 of the slave side. Arduino Pro Micro pinout:
[image]

yoution, 2019-12-14:
@mattdibi  thank you.It's ok now, I just resoldered some pins  of Pro Micros

closed 2019-12-14 by mattdibi

toindev, 2020-04-13:
Hello! It's been a while. Just for reference, my initial issue really was all about defective controllers. I finally got the time to dive into it, and mercilessly cut the controller out with a Dremel (Just the slave one). BTW, you should insist a lot more on the advantages of socketed controllers! 

Thanks again for your design and your help!

-- Written with my Redox :D