mattdibi/redox-keyboard #34: proton-c compatibility?

danielo515, 2019-07-26:
Hello and congratulations for this great project.
I was looking for a wireless alternative to my ergodox, and this looks quite awesome.
However, now that I'm going to make an "upgrade" I would love to get rid of space limitations I currently have. Proton-c is an arduino pro micro alternative that has way more space for the firmware, and it should be a drop-in replacement.
I think it still has some problems with split keyboards, but since here we only have one keyboard maybe it is not that "split" at the end.
Do you think it will be compatible?

mattdibi, 2019-07-26:
Hi,
I assume you're talking about the Redox Wireless.
Footprint-wise for sure. The code doesn't use the split functionalities so I think it should work.
I don't know of anybody who actually tried this though. You'll be the lucky beta tester if you proceed with this experiment :D

danielo515, 2019-07-27:
Yes, I was talking about the redox wireless 😄
So physically it should fit. I don't see a reason why it should not work on the software level. I mean, I'm not an expert but proton-c uses the same firmware with probably just some tweaks on the settings.

By the way, do you have any picture where I can see the receiver with the arduino attached? All the pictures I found are only showing the wireless part

mattdibi, 2019-07-27:
Here's some pics. Usually I add some female header pins to the receiver to be able to remove the Arduino in case I need it.
[image]
[image]
[image]
[image]
[image]

In case you're wondering without those female header pins the arduino + receiver assembly should look somethink like this.

[image]
[image]
[image]

Maybe a little thicker because there's the voltage regulator on the back of the receiver board.

Hope this helps.

danielo515, 2019-07-27:
Wow, that is quite a big receiver! But I like the idea of the female pins.
Thank you very much for the pics, they helped me to get an idea about how it looks like.

closed 2019-09-07 by mattdibi