mattdibi/redox-keyboard #35: No key presses working

termhn, 2019-08-08:
Hi!
I've (in theory) just finished soldering and flashing firmware for my Redox, however no keys are working on either side. I believe my flashing went well since when I plug in either side now, it is displayed as a Generic Keyboard device in windows Device Manager rather than as an Arduino Pro Micro as it had before. However, like I said, none of the keys work on either side, neither when connected individually or together with the TRRS.

I don't see any obvious soldering issues. Any ideas? Here's a couple pictures of the backs of the boards, dunno how much good that will do :p https://imgur.com/a/Duoo4bg

termhn, 2019-08-08:
Okay, I'm very perplexed. I further was debugging this issue and went thru with my multimeter and was getting the proper voltage on both sides of the switch both when open and close when they were plugged in. 

Long story short, after a reboot of my computer... it magically works :)

closed 2019-08-08 by termhn

mattdibi, 2019-08-08:
Maybe it was a driver issue? Anyway glad you fixed it!