mattdibi/redox-keyboard #36: Question mark

ddon, 2019-08-13:
Not sure where is the right place to ask, but I am enjoying my first Redox Wireless keyboard and so far, so good... slowly learning hot wo type again :) So far, the biggest problem is -- how do I type question mark? :)

mattdibi, 2019-08-14:
Hi, glad you like it! :)
Regarding the key positions you can refer to the standard layout [here](https://raw.githubusercontent.com/mattdibi/redox-keyboard/master/img/redox-layout.png), or directly in the [firmware default layout code](https://github.com/qmk/qmk_firmware/blob/547fbe769c684745195a53baf9f62730ceea804d/keyboards/redox_w/keymaps/default/keymap.c#L36).

The question mark is on the key on the left of the right shift.

ddon, 2019-08-14:
next to the right shift I have "\" button, and using it with Shift makes it "|"... 

I checked your link to firmware default layout code, and it shows that exactly there supposed to be backslash key...

[image]

mattdibi, 2019-08-15:
Well... that's awkward.

I'm sorry, apparently I made a mistake in the keymap and totally forgot about the question mark (my only excuse is that usually I use the italian layout).

I should update the keymap on the QMK firmware but it'll take some time. I would suggest you to update the keymap on your own. I would suggest to move the `BKSL` key as a momentary key under the right shift and move the `SLSH` in place of the `BSKL` key. What do you think?

ddon, 2019-08-19:
glad we found the problem... so, i understand i will need a programmer to re-program keyboards in order to update keymaps, right? 

please tell me how "momentary key" work, I am new to this :)

biggestj, 2019-08-19:
DDon, you'll need to flash that using QMK. There's a guide here: https://docs.qmk.fm/#/newbs_getting_started. 

If you follow along that guide you'll be able to get what you need, don't worry you don't need to be a programmer. It might seem really complex at first, but you don't need to know what's happening under the hood too much. Make sure you follow the guide for your operating system. 

I'll attach the default keymap (so you don't have to change anything in 'code') with slash+questionmark on their default qwerty position and I've moved the \ button to the left ctrl.
https://dsc.cloud/314c8f/default_questionmark.zip

@mattdibi I've been using the board for 4 days now, and only just now ran into the issue of not having a question mark. So from my point of view it's understandable.

biggestj, 2019-08-28:
@ddon, it has been fixed now. If you use the default layout in the toolbox and upload it to the redox you should have the fixed layout.

closed 2019-11-22 by mattdibi