mattdibi/redox-keyboard #38: Is the 4th LED on the addressable?

pixelbreaker, 2019-08-26:
@mattdibi Is it possible for me to control the state of the 4th LED on the receiver of the redox-w?

I'd like to be able to control it and light it up when CAPS lock is on, I've written the rest of the code to handle this, but my hardware/arduino knowledge is lacking here.

I'm sure it's something I need to add here
```c
#define red_led_off   PORTF |= (1<<5)
#define red_led_on    PORTF &= ~(1<<5)
#define blu_led_off   PORTF |= (1<<4)
#define blu_led_on    PORTF &= ~(1<<4)
#define grn_led_off   PORTD |= (1<<1)
#define grn_led_on    PORTD &= ~(1<<1)
```

pixelbreaker, 2019-08-26:
I've figured it out

https://github.com/pixelbreaker/qmk_firmware/blob/master/keyboards/redox_w/keymaps/pixelbreaker/keymap.c#L38

mattdibi, 2019-08-27:
I see you figured it out. For further informations refer to https://github.com/mattdibi/redox-keyboard/issues/24 .

closed 2019-08-27 by mattdibi