mattdibi/redox-keyboard #45: Issues Flashining Pro Micro

ktrieun, 2019-11-22:
Trying to flash the pro micro, but am running into an issue. I'm unable to get the pro micro to reset. When I press the reset button, the pro micro does flash a very dim red light across two LED

This is the pro micro that I'm using. Does the fact that it is a 5V version matter?
https://www.aliexpress.com/item/32768308647.html?spm=a2g0s.9042311.0.0.7aa64c4dq0JLOp

mattdibi, 2019-11-22:
Hi,
the Pro Micro version is correct. Double check for possible shorts on the Pro Micro pins. If all else fails (and by no means this is a long-term solution) you can short the reset pin to ground using some tweezers.

ktrieun, 2019-11-25:
I just got new Pro Micro boards and I tried shorting the reset and ground pins, but it still isn't being recognised by QMK when I try to flash it. the

closed 2019-11-25 by ktrieun