mattdibi/redox-keyboard #46: Rotary Encoder on redox wireless?

Slavfot, 2019-12-01:
Hi!
Would it be possible to add a rotary encoder to the wireless redox?
How would you wire it to the wireless transmitter and is it hard to fix the code to handle a rotary encoder?

Best Regards /Anton #

mattdibi, 2019-12-01:
Possible duplicate of #40 .

Hardware wise you would need to:
- Modify the case to fit an encoder (I don't know if it can fit between case and PCB)
- Wire the encoder directly to the YJ-14015 (handwiring is the only way, the PCB doesn't support this)

Software wise:
- Write the code to read the encoder
- Come up with a way to send it through the gazell protocol
- Receive it on the arduino side and map it to QMK (remember the YJ-14015 don't use QMK)

I would say it isn't easy. You're welcome to modify the available files, I cannot right now due to lack of free time.

Hope this helps.

closed 2019-12-01 by mattdibi

Slavfot, 2019-12-01:
Thanks for the reply!
Is there anywhere you can point me for information on how to learn to under stand the YJ-14015 and how to write code for it?

mattdibi, 2019-12-02:
Take a look at the [firmware](https://github.com/mattdibi/redox-w-firmware) and the Nordic Semiconductor SDK [documentation](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Fgroup__gzll__02__api.html&anchor=ga8be12e26c93ee706e89b0ec2a491c6ab).

LukeAndersonTrocme, 2022-07-13:
@Slavfot did you manage to figure this out? I'm wanting to make something similar to this except using the ATMEGA32U4 and ideally QMK.

Jus to be sure I understood @mattdibi's reply, is it the physical shape of the PCB thats incompatible or is it the PCB wiring not compatible with the signal generated by the rotary encoder?

Kudos for the sharing this great open source project!

PhiBabin, 2023-04-27:
Last year, I made a version of the redox wireless with a rotary encoder. I made a modified version of the Redox which has the option of replacing one of the key for an encoder. I updated the firmware to support sending the encoder change over the wireless connection. I also update the 3D printed case for it:
[image]

I kind of forgot to publish everything on Github, so if there are people interested in that mod, please let me know.

TweetyDaBird, 2023-08-14:
Very much interested. I can easily make the PCB changes myself, but I'm curious about the code changes.

PhiBabin, 2023-10-08:
I made a repository with all the firmware, PCB and 3D printed files for the rotary encoder support:  https://github.com/PhiBabin/redox-w-encoder