mattdibi/redox-keyboard #47: Redox-w firmware flash help

l-ll, 2019-12-20:
- In a terminal show `nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints`.
- another terminal step by
```
echo reset halt | telnet localhost 4444
echo nrf51 mass_erase | telnet localhost 4444
echo reset halt | telnet localhost 4444
echo flash write_image `readlink -f path/to/receiver.hex` | telnet localhost 4444
echo reset | telnet localhost 4444
```
- same step for left and right hand.
- then flash `qmk_redox_w_default.hex` to pro micro.

- YJ-14015 and battery has same voltage (near 3v)

also repeat flash step with another 3 board, only one board can working after first time flash :(
I am very confused why only the first time working.

mattdibi, 2019-12-21:
Hi,

Just to be sure I got the problem right: only one board is working out of three, which is the one that was flashed first?

If so, would you mind attaching the output from the openocd terminal following the telnet commands?

Also in the latest version of Redox-W firmware you can find three [programming scripts](https://github.com/mattdibi/redox-w-firmware#automatic-make-and-programming-scripts) that will make your life easier.

l-ll, 2019-12-22:
same flash step
but now it's working...

closed 2019-12-22 by l-ll