mattdibi/redox-keyboard #48: Single row on left hand not working, how to troubleshoot?

harshitgoel96, 2020-01-11:
Hello.

The entire keyboard is working fine except one single row. I do not see any issue on MCU soldering as all pins are visible joined. Please let me know how can I trouble shoot this issue.

mattdibi, 2020-01-12:
Hello.

Double check soldering. Visual inspection is not sufficient to determine good soldering. Use a multimeter in continuity mode.

Double check PCB trace for an interrupted line (highly unlikely).

harshitgoel96, 2020-01-12:
@mattdibi i was able to get the work with reflowing the solder. it has resolved the issue for me. however i have some issue with  the dongle range. does the board to be in line of site? i some times see the issue with slow/dropped key registers. moving the keyboard a bit solves  the issue.

mattdibi, 2020-01-12:
@harshitgoel96 Happy to hear that!

Range depends heavily on environmental factors: an environment with crowded 2.4GHz band (a lot of wireless devices) can be a contributor. Aluminium case can negatively impact the range. Ensure you're using the latest version of the Redox firmware for the [latest range/stability fixes](https://github.com/mattdibi/redox-w-firmware/pull/7).

closed 2020-01-12 by mattdibi