mattdibi/redox-keyboard #49: left RGB lights not working

solidgriever, 2020-01-29:
Hi, I build a rev1 redox with RGB and right side RGB does not work.
It is already soldered and it will be a pain to desolder the micros and the switches. Hope someone has solved this issue. 
(all switches work OK see 3) and all RGB leds turn on see 2) )
what I tried:
1) flashed redox_rev1_default_RGB.hex(precompiled mattdibi/redox-keyboard):
       -left side: OK both keystroke detection and RGB
       -right side: no keystroke detection no RGB
2) flashed redox_rev1_default.hex (precompiled  mattdibi/redox-keyboard)
       -left side: OK detecting heystrokes and OK RGB 
       -right side: no keystroke detection but OK RGB
3) compiled and flashed from source default keymap(make redox/rev1:default)
        -left side: OK keystroke detection & OK RGB
        -right side: OK keystork detection & no RGB   (sometimes i get the 1st led on set to green)      

if i try to set comunication to I2C i get very slow reaction on both RGB and keystorkes

mattdibi, 2020-01-30:
Hi,

I would double check the wiring of the RGB strip, is it correctly wired as seen [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox#rgb-underglow)?

solidgriever, 2020-01-30:
thx for the replay. I think the wiering is ok because if i flash your precompiled default.hex the rgb strips behave as I expect: I can toggle both and 
[image]
change mode from the left side.

mattdibi, 2020-02-01:
Hi,

I've noticed you soldered the 4.7k resistors near the PJ-320A connectors, which is wrong. The resistors are needed for the I2C connection. The RGB underglow works by occupying one wire of the TRRS connection for transmitting the data between the two halves, thus preventing us to use the I2C connection (which uses two wires).

So:
- Remove the two resistors.
- Be sure to use the **serial** communication instead of I2C (see [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox#rgb-underglow-firmware))
- Be sure to define the correct **master hand** (see [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox#rgb-underglow-firmware)) .

Hope this helps.

solidgriever, 2020-02-01:
Yes i soldered the resistors as a test just before i took the picture and it did make no difference.
Anyway, i managed to make it run as intended. 
I think the problem is the right PJ-320A jack.
What i did was insert the cable only ~80% in the jack.

closed 2020-02-01 by solidgriever