mattdibi/redox-keyboard #50: How to flash the keyboard?

MarkuBu, 2020-02-01:
Hi,
I have no idea how to flash the keyboard with a new firmware (I'm using Ubuntu 18.04). I can't find /dev/ttyACM0 device or similar. 

I have the redox keyboard with RGB Led. The README says that I have to use the serial port, but I don't know how to set the keyboard to program mode. 

thanks for help
Markus

MarkuBu, 2020-02-01:
I know a bit more now. But I have another issue. if I run make with avrdude I get the message

Waiting for /dev/ttyACM0 to become writable

If I run with sudo I get tonns of error messages like

avrdude: error: programmer did not respond to command: set addr
Reading | #################################                  | 66% 0.24savrdude: error: programmer did not respond to command: set addr

I hope I don't ruin my keyboard

The documentation on this page could be better

MarkuBu, 2020-02-01:
Do I have to add something to /etc/udev/rules.d?

The documentation here is horrible. Please add more information. It was easier to flash my Ergodox compatible keybord

mattdibi, 2020-02-02:
Hi,

> Waiting for /dev/ttyACM0 to become writable

You need to push the reset button highlighted in the following picture.

[image]

For more information about building see the [QMK documentation](https://docs.qmk.fm/#/faq_build).

>The documentation here is horrible. Please add more information. It was easier to flash my Ergodox compatible keybord

This is an open source project, you're welcome to expand and contribute the documentation you feel needs more details.

MarkuBu, 2020-02-02:
Hi, 
meanwhile I get the answer

https://github.com/qmk/qmk_firmware/issues/8060

Sorry about the rant, but I was a bit upset. My keyboard is working again, but I haven't found this information in the documentation. Maybe you can add this in some way. As you can see I'm not the only one with this issue. 

My first impression of the Redox: I'm glad that I did not buy the Ergodox because I think that the thumb cluster is to far away for relaxed work. I just miss the index keys on the keycaps. 

Thanks for that great keyboard

closed 2020-02-02 by MarkuBu

mattdibi, 2020-02-02:
Hi,

don't worry, I understand. I'll update the documentation with this information ASAP. Thank you for taking the time to share the solution you found.

Have fun with your keyboard!

MarkuBu, 2020-02-02:
If I have some experience with this keyboard maybe I will help with some contributions, if you want. But first I need to learn more about it.