mattdibi/redox-keyboard #51: Keycaps

sluedtke, 2020-02-12:
Hi there, 

I am struggling with the size for the keycaps  and where to get them. 

I see from the documentation that sizes of 1, 1.25 and 1.5 are required. Do I have to use those exact sizes? Where are they located exactly. It looks like the 1.25 are the outer columns on each half but I am not sure. The 1.5u are located at the thumb cluster?

Does anyone has recommendations where to buy a decent set?

I am pretty new to this "businesses" ;-)

Mishkun, 2020-02-13:
You can use smaller keycaps if you struggle to find right sizes. But this would not be so aestheticaly pleasing.
For newbies I can recommend to go on aliexpress and find some keycap sets there. They are pretty cheap and some of them are good looking (although not on the peak of keycap fashion)
And forthese questions better ask on mechanical keyboards subreddit.
As for location, you guessed right.

sluedtke, 2020-02-13:
Thanks, I will check that out.

closed 2020-02-13 by sluedtke

danielo515, 2020-07-24:
I think it will be nice to link to some known keycaps that we know works nice on the main readme.