mattdibi/redox-keyboard #53: Buttons don't work or output character many times

charafau, 2020-02-18:
Hi! I soldered my first redox keyboard with PCBs from falbatech. Unfortunetly I needed to desolder one controller and solder two buttons under it because I forgot to solder them. 

Now one half's buttons don't work or output character many times, even if I only connect one half through mini-usb. Any debugging tips, hints? Should I buy one more pcb, controller and buttons ?

Thanks!

mattdibi, 2020-02-18:
Hi,

Before ordering new PCBs double check your soldering. Looks like there might be a naughty joint somewhere.

Hope this helps!

DickThomas, 2020-07-28:
i had a similar issue to this. 

turns out I had a blob of solder on the 5v/3v j1 on the arduino.

I hope this is of help.

charafau, 2020-08-09:
broken controller, thank you all for help - closing

closed 2020-08-09 by charafau