mattdibi/redox-keyboard #55: Redox Wireless Interference issues

Slavfot, 2020-03-18:
Hi! 
I just built a redox wireless and flashed it and got it working but i have trouble using it due to interference.
Especially at home in the apartment, if i press a button it may (if lucky) send a keypress or it may send multiple keypress even though i just pressed once. It does not matter if i have the reciever 1cm from the keyboard or 50cm.
I have also tested it at work when i was alone and it worked better, but when my collegues started working i got so much interference that the keyboard was unusable. 
How do i reprogram the reciever and transmitter to get less interference?
I read that you can modify in the gazell firmware and modify the default channel table /components/properitary_rf/gzll/nrf_gzll_constants.h.
If i modify (i don't know what channels to use instead) do i need to compile a new hex? (i don't know how to compile a new hex but i am willing to learn)
Thanks!

Slavfot, 2020-03-18:
I also found these lines in main.c that i suspect that i can add number of channels to?:
#ifdef COMPILE_LEFT
static uint8_t channel_table[3]={4, 42, 77};
#endif
#ifdef COMPILE_RIGHT
static uint8_t channel_table[3]={25, 63, 33};
#endif

Slavfot, 2020-03-21:
Ok, i got it to work now...
First I learned how to make new hexes. (I'm not a programmer, just a really stubborn maker)
Then i read alot in the nordic pages about Gazell link layer trying to understand things.
I tested several things including adding channels and changing channels.

That did not help.
The thing that made it work finally was to remove these rows of code:
    nrf_gzll_set_timeslots_per_channel(4);
    nrf_gzll_set_channel_table(channel_table,6);
    nrf_gzll_set_datarate(NRF_GZLL_DATARATE_1MBIT);
    nrf_gzll_set_timeslot_period(900);

I read an other issue where i beleive you and LucidityCrash added this code.

I'm not sure if it was the channel table, the datarate or the timeslot that messed it up for me.
I will make some more tests with each option later to see which line of code that messed with my keyboard.

mattdibi, 2020-03-23:
Hi Slavfot,

sorry for the late response. I'm having a hard time answering to the issues here right now.
I'm glad you found a way around your problem, let me know the results of your tests.

Cheers,
Matt

AltusBaard, 2020-03-30:
@Slavfot I am having the same issue, just to confirm, you updated only the keyboard firmware, not the receiver?

Slavfot, 2020-03-30:
> @Slavfot I am having the same issue, just to confirm, you updated only the keyboard firmware, not the receiver?

I updated both the receiver and the keyboards.
I can upload my hex files later today.

AltusBaard, 2020-03-30:
That would be great, I have only found some of the entries on the receiver and am worried of breaking things even more

Slavfot, 2020-03-30:
@AltusBaard 
Here are the hex to the Nrf51128 modules. They are made with the edits i mentioned above.
[precompiled fix.zip](https://github.com/mattdibi/redox-keyboard/files/4404686/precompiled.fix.zip)

AltusBaard, 2020-04-01:
Thanks, uploaded them and testing, seems to be working.

Also had a low battery in one, think that also might have had an impact on it.

Slavfot, 2020-04-01:
Great to hear that it worked!
If you had low battery on one half. Then only that half should have been acting weird. Since each half is talking individually to the reciever.
So if you had issues on the keyboard that had good battery that are fixed now, then the firmware update is the thing that fixed it.
Atleast that's what I think, please correct me if I'm wrong.

AltusBaard, 2020-04-02:
Seems I still  have an issue, it does feel better though, a lot less problems.

Will try see if I cant see anything else that I am missing. But thanks for helping anyway.

Pastitas, 2020-04-13:
I've been trying the modded version of the firmware and i think that it's a tiny little bit better (might be placebo) but still get this happening every now and then.

I think what's necesary is to do some debugging to figure out who the culprit is, something is mentioned to that end in these two: https://github.com/mattdibi/redox-w-firmware/issues/2 https://github.com/mattdibi/redox-w-firmware/issues/3

SimonMerrett, 2020-08-16:
Hi all, I would like to thank @mattdibi for this firmware but when I was just testing out a modified version of it for the [aerodox](https://hackaday.io/project/171535-aerodox) keyboard, I had *terrible* chatter issues. Without being familiar with NRF51 debug or QMK, I was combing through the code looking for issues I had introduced with my port/edit. I also switched out my Otemu switches for legit Cherry MX blues to make sure it wasn't a dud set of switches. The only thing which worked was:

> The thing that made it work finally was to remove these rows of code:
> nrf_gzll_set_timeslots_per_channel(4);
> nrf_gzll_set_channel_table(channel_table,6);
> nrf_gzll_set_datarate(NRF_GZLL_DATARATE_1MBIT);
> nrf_gzll_set_timeslot_period(900);

To be clear, I left in `nrf_gzll_init(NRF_GZLL_MODE_HOST);` in the receiver code (there's no `nrf_gzll_set_timeslots_per_channel();` in the receiver code - should there be?).

The difference in reliability/sensitivity is night and day - I would recommend a thorough review of this part of the Redox codebase by those who are more familiar with it as I think it could be causing significant issues if my experience is anything to go by.

Thanks for your persistence and sharing @Slavfot

harshitgoel96, 2020-10-01:
I am having similar issues with redox-w. I have used it for 10hr, and I have interface, lag, and key drop issues on the right hand. sometime some keys of the keyboards stop working. 
i have use fresh batteries so its not a battery issue. 

is there a pre compiled for transmitter and receiver ?

SimonMerrett, 2020-10-01:
I have also noticed things getting gradually worse. Lag and keydrop are more noticeable than before but I can't tell if it is progressively worse or I am just more likely to notice it now that I am used to the keyboard. Alt+tab is the one which goes most wrong, as it just records tab on its own and that often does something different to a window switch. To be clear, I am using an aerodox which is a derivative of the Redox-w control system but not the PCBs. I have tried power cycling all boards and my l/r halves are 2.9V each so should be fine for the nrf's voltage (using a pair of AA nimh rechargables on each half, so internal resistance isn't an issue at that voltage). 

Another issue that is related to this is the matrix scanning doesn't allow e.g. shift+key if they are same row - I have to use the opposite shift key and can't do one handed for some keyboard shortcuts. It could be something in my adapted fw but I wondered if this was an issue with the stock hw/fw combo.

@harshitgoel96 I think there is Redox-w precompiled fw - I will link if I find it but keep looking!

Additional note - I have just tried moving my keyboard further away from my Surface Pro 2 and it seems to slightly improve the reliability of the input. Do you have any possible 2.4GHz interference from wifi or bluetooth radios that you could move away from or turn off. It was definitely something I noticed early on in using my keyboard. Make sure you have reasonable line of site (and perhaps not too close) to the receiver and that it is also away from 2.4GHz sources.

SimonMerrett, 2020-10-01:
@harshitgoel96 https://github.com/mattdibi/redox-w-firmware/tree/master/precompiled

harshitgoel96, 2020-10-01:
[image] this my regular setup. I have wifi 5ft away from keyboard.

I have bluetooth headset, but I have problems even when they are off. Mobiles bluetooth and wifi could be an issue?

Is there any way to reduce the interference ?

SimonMerrett, 2020-10-01:
Could you try using a longer usb cable on the receiver? Doing some more testing now, the further from my Surface that the keyboard is, the better the reliability is. Do you have wifi or bt enabled on the laptop? That would seem like the main source of interference to me. And also I can't quite remember where the antennas are in the Redox hw but you could have the left half of the keyboard and the laptop blocking line of sight to the receiver's antenna. Perhaps try the longer USB cable and moving the receiver around (higher and away from the laptop).

harshitgoel96, 2020-10-02:
so turning off the laptop and mobile bluetooth does help. But i am curious why this happens ? Is it hardware related issue like the antenna of the BT chip or more of a firmware issue ? Other BT dongle keyboards do not have drops like this, but they are not split either. so just curious. 

adding antenna lines on pcb would help ?

harshitgoel96, 2020-10-05:
Ok, I did some more playing around, and found that receiver when placed in direct line of sight works good. It makes me wonder is it the antenna of yj-14015 is too weak ? 

The yj-14015 does not have option for external antenna. 

Does anyone know if mitosis has the same interference issue ?

SimonMerrett, 2020-10-11:
Still got issues here. The removal of my phone to a metre away helps. But I'm starting to think about another wireless solution. The redox-w does have different radio settings, as per @Slavfot's report above (I commented out the code he mentions and this improved my performance) but did the timeslots perform a needed function or something? They don't appear in the mitosis `main.c`

I couldn't say if there are issues with the mitosis keyboard. BTW @harshitgoel96 this is using the gazelle protocol, not bluetooth. I don't know much about bluetooth and I know less about gazelle so I would value any insight @Slavfot or others have gleaned from their research. Perhaps a reduced datarate (down from 1mbit), a different number of retries (100) or action on failed tx would help.  The redox-w code also differs from the mitosis wrt the debounce code, so maybe that's worth a look too.

Pastitas, 2020-10-12:
I commented [here](https://github.com/mattdibi/redox-w-firmware/issues/12)
that somebody optimized the code from the original mitosis keyboard, it seems, based on what they say on [this issue](https://github.com/reversebias/mitosis/issues/1) that the keyboard gets more reliable and longer battery life. Based on what you are saying the shorter battery life could be related to constant resending due to interference, wich said fork is suposed to help with. I want to get into it an try an port the changes over and it does not seem too complex to do, but i don't have much time lately for this.

harshitgoel96, 2020-10-14:
Joric modified the mitosis firmware to add native bluetooth capabilities to the keyboard. I am not technical enough to port it to Redox. Can someone try to do this and check if this has better communication then Gazell protocol.

https://github.com/joric/mitosis

SimonMerrett, 2020-10-18:
> Can someone try to do this

I had a look but the problem I have is that I have a different layout with aerodox so even if I managed to make it work, it wouldn't help you that much. I'm going to have a look at the link suggested by @Pastitas but don't hold your breath...

Pastitas, 2020-10-18:
> > Can someone try to do this
> 
> I had a look but the problem I have is that I have a different layout with aerodox so even if I managed to make it work, it wouldn't help you that much. I'm going to have a look at the link suggested by @Pastitas but don't hold your breath...

If you manage that much i am willing to take it to the original redox layout

SimonMerrett, 2020-10-19:
I haven't tried yet with the firmware changes but I have just tried moving the receiver well away from anything - about half a metre to a metre away from the computer and the keyboard. The reception is much better now (definitely still not reliable enough for my taste) so part of the problem may be that the keyboard is too close to the receiver and is swamping it with energy. Even on 0dB it might be. So if you do try to increase the power, bear in mind that unless there's some adaptable attenuation going on in the nrf analog front end, it might be a case of more range between keyboard and receiver also improves things.

SimonMerrett, 2020-10-21:
I have just 'upgraded' to 4dBm transmit power on the keyboard transmitters and set them to 200 retries instead of 100. There's still some potential to handle e.g. `nrf_gzll_device_tx_failed()` with something to resend unsuccessful packets. I also wonder if there's any merit in studying  the effects of the rx fifo pipe flush. 

It's too early to tell how much difference the power and retries increase has made but I will update when I have spent some more time with the keyboard in this mode. I can say that it does appear to work more fluidly, so you have nothing to lose (apart from battery life) by trying it yourself.

harshitgoel96, 2020-10-21:
> I have just 'upgraded' to 4dBm transmit power on the keyboard transmitters and set them to 200 retries instead of 100. There's still some potential to handle e.g. `nrf_gzll_device_tx_failed()` with [...]

can you share you compiled hex for transmitters ? i will add them to my keyboard and test it out.
also it will be on keyboard or on dongle or both ?

SimonMerrett, 2020-10-21:
> can you share you compiled hex for transmitters ? i will add them to my keyboard and test it out.
> also it will be on keyboard or on dongle or both ?

Sorry, as mentioned above I'm using this on a custom layout which is not directly compatible with the redox-w. I will say that I also felt that just changing the power was a little unsatisfactory to I went back and looked at the code that @LucidityCrash added and @Slavfot (and I, on my board) removed to get an improvement in the connection between the keyboard and receiver.

So Lucidity crash was following [advice from the nordic support team](https://devzone.nordicsemi.com/f/nordic-q-a/4427/advice-for-optimizing-multiple-device-gazell-system) when they suggested using the channel hopping mode for two transmitters to one receiver. Compared to how I think Gazell is supposed to work, we're using it in a fairly blunt manner. I noticed something in that advice that I don't think LucidityCrash applied in their patch: 
> You must also keep the PTXs the time slot twice the length of the PRX

So this means we need to `nrf_gzll_set_timeslot_period(1800);` in the receiver, not 900. Also, I then went to look at the Gazell user guide and see that they have some parameters about whether to use the previous successful channel on the tx and also the sync method used. So I've added those, dropped the bitrate to 1Mbps, kept the power at 4dBm, and will see how things go. So far, I can say it seems way better than when I had channel hopping set up as per the existing repo. But I think there may still be a couple of missed keypresses and I couldn't say it's better than omitting the channel hopping and just leaving the power up high!

As I said, I won't share the whole source for my different columns/rows layout and as I'm editing my version I won't provide precompiled redox-w binaries. But I will post snippets of code for you to insert and compile yourself in a future comment (from my other machine).

SimonMerrett, 2020-10-21:
In receiver `main.c`
```
// Define channel hoping parameters
#define channel_table_size 6
#define timeslots_per_channel 2
#define GZLL_RX_PERIOD 1800 // supports 1Mbps

static uint8_t channel_table[channel_table_size]={4, 25, 42, 63, 77, 33};
```
also
```
    // Initialize Gazell
    nrf_gzll_init(NRF_GZLL_MODE_HOST);

    nrf_gzll_set_timeslots_per_channel(timeslots_per_channel);
    nrf_gzll_set_channel_table(channel_table,channel_table_size);
    nrf_gzll_set_datarate(NRF_GZLL_DATARATE_1MBIT);
    nrf_gzll_set_timeslot_period(GZLL_RX_PERIOD);
```
then in keyboard `main.c`
```
#define GZLL_RX_PERIOD 1800 // supports 1Mbps

// Define channel hoping parameters
#define channel_table_size 3
#define timeslots_per_channel 2

#ifdef COMPILE_LEFT
static uint8_t channel_table[3]={4, 42, 77};
#endif
#ifdef COMPILE_RIGHT
static uint8_t channel_table[3]={25, 63, 33};
#endif
```
also
```
// Initialize Gazell
nrf_gzll_init(NRF_GZLL_MODE_DEVICE);

// Attempt sending every packet up to 200 times
nrf_gzll_set_max_tx_attempts(200); // added to tranmitter to see if the link is improved
nrf_gzll_set_tx_power(NRF_GZLL_TX_POWER_4_DBM); // added to tranmitter to see if the link is improved 
nrf_gzll_set_timeslots_per_channel(timeslots_per_channel);
nrf_gzll_set_channel_table(channel_table,channel_table_size);
nrf_gzll_set_datarate(NRF_GZLL_DATARATE_1MBIT);
nrf_gzll_set_timeslot_period(GZLL_RX_PERIOD / 2);
nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*timeslots_per_channel);
nrf_gzll_set_device_channel_selection_policy(NRF_GZLL_DEVICE_CHANNEL_SELECTION_POLICY_USE_SUCCESSFUL);
```

YMMV. Given that I seem to have repeat offenders in which key I am pressing (r, p - but not redox matrix) then I am as inclined to think that the missed keys are as much the debounce and key reading as it is a wireless comms issue. Good luck.

LucidityCrash, 2020-10-26:
> So Lucidity crash was following [advice from the nordic support team](https://devzone.nordicsemi.com/f/nordic-q-a/4427/advice-for-optimizing-multiple-device-gazell-system) when they suggested using [...]

Damn it ... that was a mistake ... I could have sworn I'd done that :)

SimonMerrett, 2020-10-26:
Not to worry, and from my testing so far, all it does is makes the channel hopping on par with the non-channel hopping version. I still get missed keypresses, sadly. I know I have different hardware to everyone else but I may look at the matrix scanning side of things next.

danielo515, 2020-12-11:
Hey.
I am also experiencing problems on reliability, but they are very intermittent. In some situations, I thought it was a battery problem so I just replaced the battery on the half that was giving me problems, but then after just a couple of weeks, I had the same problems again, and always on the left half.
Moving the receiver around helps, but I'm not sure about which positions are best. Right now is in the middle of the two halves, like 10 centimeters away of each other.
I also noticed that turning wifi on or having the phone close disturbs it, but my bluetooth mouse doesn't seem to cause any trouble for example.

Is there any known configuration that improves the reliability?
Oh, and do you have any guide to flash each halve? Never did it before.

LucidityCrash, 2020-12-11:
> Another issue that is related to this is the matrix scanning doesn't allow e.g. shift+key if they are same row - I have to use the opposite shift key and can't do one handed for some keyboard [...]

Just noticed this ... it must be your adapted FW as my shift Z (same row same board) works just fine :)

LucidityCrash, 2020-12-11:
> > >
> 
> Damn it ... that was a mistake ... I could have sworn I'd done that :)

Hang on a sec, from that post : " You must also keep the PTXs the time slot twice the length of the PRX "
You've made the RX twice the TX haven't you ?

SimonMerrett, 2020-12-11:
> it must be your adapted FW

Brilliant, thanks for the confirmation - now I can go and feel like time spent addressing this is worth doing. 

As an update, I have been using the "correct channel hopping" I mentioned above but the performance is still bad. However, I still think that this could be a function of the debounce settings I have (I think I went down to 5) and number of retries. Often I will have to press a key two or three times before it is registered and when it finally registers I get three of that key appear onscreen. I also get double-typed keys when I only meant to press once, hence my thought that key debounce is not effective with the 5ms setting I'm using (I think - need to check). I reduced the debounce period because I thought that might be contributing to the failure for some key presses to register (mistaking a press and release too quickly for no press at all). 

I plan to at least increase the debounce and maybe play with some other things soon to try and make this a more viable keyboard. At the moment I am enjoying cheap lenovo membranes more!

SimonMerrett, 2020-12-11:
> You've made the RX twice the TX haven't you ?

Yes! it looks like I have. Will swap that round and try it out.

SimonMerrett, 2020-12-11:
@LucidityCrash I think I got confused by [this ](https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v4.x.x/doc/html/group__gzll__02__user__guide.html) when you scroll to the bottom:
```
/* On Host and Device */
timeslots_per_channel = 2;
channel_table_size = 3;
nrf_gzll_set_timeslot_period(GZLL_RX_PERIOD / 2);
nrf_gzll_set_channel_table(my_channel_table, channel_table_size);
nrf_gzll_set_timeslots_per_channel(timeslots_per_channel);
/* On the Device */
nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*timeslots_per_channel);
nrf_gzll_set_device_channel_selection_policy(NRF_GZLL_DEVICE_CHANNEL_SELECTION_POLICY_USE_SUCCESSFUL)
```

LucidityCrash, 2020-12-11:
yeah this : 
`nrf_gzll_set_timeslot_period(GZLL_RX_PERIOD / 2);`
is really confusing the way I'm reading it and the doc for the function https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v4.x.x/doc/html/group__gzll__02__api.html#ga3347094a64c07793c773c4fdf4bc4bff

is that GZLL_RX_PERIOD is the time for 2 attempts but nrf_gzll_set_timeslot_period() sets the time for a single attempt

SimonMerrett, 2020-12-11:
I think what I implemented was "correct" frequency hopping but for a single Tx/Rx pair, rather than a 2*Tx/Rx trio. If the boards are 1800 and Rx is 900, I think that should work because the Tx should come back round to channel x1 after channel x3 at the same time as Rx comes back to channel x1 after cycling through x1, y1, x2, y2, x3, y3...

karosc, 2020-12-11:
> Hey.
> I am also experiencing problems on reliability, but they are very intermittent. In some situations, I thought it was a battery problem so I just replaced the battery on the half that was giving [...]

Very weird, I experience the exact same issue. Either my keypresses are not recognized or they come in way after I press the key. Sometimes they come in repeating indefinitely until I try to press another key, which interrupts the repeat. This is always only on the left half. The problem can be resolved for a while by taking out and reinserting the battery (effectively turning the transmitter off and on, but the problem always comes back.

SimonMerrett, 2020-12-11:
> Either my keypresses are not recognized or they come in way after I press the key.

Very similar to (one of) my issue(s):

> Often I will have to press a key two or three times before it is registered and when it finally registers I get three of that key appear onscreen. I also get double-typed keys when I only meant to press once

LucidityCrash, 2020-12-11:
> Very weird, I experience the exact same issue. Either my keypresses are not recognized or they come in way after I press the key. Sometimes they come in repeating indefinitely until I try to press [...]

Very strange ... are you using the Firmware in the repo or from the supplier, or did you build yourself ?
I've been using this for over a year and had a few issues at the beginning but some related to battery levels I believe but other than that it has been pretty good.

karosc, 2020-12-11:
I modified the firmware to allow for an additional column for my build. So somewhat customized. I also tried modifying the channel hopping code according to SimonMerretts' comments and also tried removing it entirely; neither really solved my problem. Interestingly, I find the issue comes up mostly after my PC is been off/sleeping for a while.

SimonMerrett, 2020-12-11:
As I'm part of the problem here I thought I'd change the timings and see what happens. Both halves of my keyboard are now 
```
nrf_gzll_set_timeslot_period(GZLL_RX_PERIOD);
```
with the receiver left as ```GZLL_RX_PERIOD / 2```
So far, so good. I will update further once I have had a bit longer to see if there are any more issues.

SimonMerrett, 2020-12-11:
I turned my debounce to 10, from 5 as per original repo, and I am finding far fewer repeated typped keys (although some slip through as you can see!). I may shift this up again, as I am not a particularly fast (= slow) typist and my outemu switches are at the budget end of things and are probably not helping in terms of debounce.

LucidityCrash, 2020-12-12:
> Often I will have to press a key two or three times before it is registered and when it finally registers I get three of that key appear onscreen

Just a silly question ... what version of the nRF5 SDK are you building against ?  I tried to build against a newer version that the v11 one specified and I got some really weird behavior similar to that.

SimonMerrett, 2020-12-12:
> what version of the nRF5 SDK are you building against ? I

Not a silly question. I had to go and check though! Yes, version 11.0.0 release date: Week 10 2016

karosc, 2020-12-15:
Currently, the host has a timeslot period of 900ms (GZLL_RX_PERIOD / 2) and two timeslots per channel, making for a total 1800ms of listening on a channel before moving onto the next . The devices have a timeslot of 1800ms and two timeslots per channel, so they try transmit on each channel for 3600ms before moving onto the next; this scheme keeps the channel tables between the host and device in-sync. However, the [nordic docs](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v11.0.0/gzll_02_user_guide.html#timeslot) state that:

> This Host [period] must be greater than the time required committing 2 full transmission attempts on the Device (including ACK wait time).

If 2 full transmission attempts on the devices happen in 3600ms, then the host must be listing on a channel for that duration, but we currently have it only listening for 1800ms. I think this is our problem. The nordic docs would suggest that you increase the GZLL_RX_PERIOD on the host so it is listening for 3600ms, but the problem with that is our host has twice as many channels and our device and host would fall out of sync with respect to which channel they are transmitting and listing on. 

We may be able to solve this by setting both host and device to have timeslot periods of 900ms, but then setting the device to have 4 timeslots per channel. This way the host listens on each channel for 1800ms, which is large enough to capture two transmission attempts from the device (2 times the 900ms timeslot period). But because we set 4 timeslots per channel on the device, it stays on a channel for twice as long as the host before moving on to the next one in the list, allowing the host and device channel heartbeat to stay in sync.

This took me a while to realize, and I know I am not explaining very well here. Maybe diagram it to make things easier.

SimonMerrett, 2020-12-15:
@karosc, thanks for persisting with this. I'm sure we will work it out! I have read a bit more on the Nordic docs, trying to better understand timeslots, channel hopping, transmit durations, ack durations, retransmit behaviour etc, etc.

> This Host [period] must be greater than the time required committing 2 full transmission attempts on the Device (including ACK wait time).

Understanding the reason for this is key, which I am only starting to do! The reason is because sync is never perfect and the Device may begin a transmission before the Host is ready (still on previous channel for example), in which case reception will not happen and no ack will be sent. By having the "Host [period]" greater than the time for 2 full transmission attempts, it guarantees that a subsequent transmission will be fully captured if the Host is lagging the Device (in terms of sync). Well that's my latest understanding!

we know `nrf_gzll_set_timeslots_per_channel()` only applies to in sync device and host.
We also know that `nrf_gzll_set_timeslot_period()` should be >= 600us for 2 MBit/s or >= 900us for 1MBit/s. I cannot yet tell whether this allows for an ack listening period after the transmission period. 

After the transmit (and hopefully time to listen for and receive an ack), the transmitter waits until the next timeslot to either retransmit the same packet (if no ack received, until ack received or max retries is reached), transmit the next packet (if ack was received) or change channel (if timeslots per channel has been reached) and then carry out a retransmit or transmit next packet (as per the first two cases). 

As we do not know how long each transmission actually lasts, except to use as a guide the minimum duration from the`nrf_gzll_set_timeslot_period()` documentation. So let's think of `nrf_gzll_set_timeslot_period()` as the period between the start of transmission, rather than the time taken to complete a transmission (which I was thinking before).

If we have two transmit attempts, 1800 between the start of each, we get 3600 us before a channel change on the Device. But on the host side it is only listening for 2x 900 = 1800, so we are hoping that the beginning of one of the two transmissions for a given channel is in the first 900 us of that total 1800 us receive period (because there may not be time to receive if it begins in the last 900 us?). This is definitely wasting a transmission. I think increasing `nrf_gzll_set_timeslots_per_channel()` to 4  and reducing `nrf_gzll_set_timeslot_period()` to 900 from 1800 on the Device is a good idea.

I also wonder if there are too many retries allowed on my Device, which could potentially be blocking the Tx FIFO so subsequent presses are being ignored (FIFO full will not accept more data). So I may try reducing those.

further, moving away from in-sync Device/Host pairs, we should consider the sync process. I wonder if our two-Device-one-Host systems are resulting in the two Devices wrestling the sync of the Host each time they transmit alternately. So we could find the Host out of sync with a Device more often than in a one-Device-one-Host system, but I'm speculating here. I noticed that @karosc increased his `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*(timeslots_per_channel+1))` on the receiver and keyboard. My current code is for `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*(timeslots_per_channel))`. However, the Nordic docs say:

> This value should be set so that the Device transmits on one channel while the Host goes through a full channel rotation, i.e., channel_table_size*timeslots_per_channel. This ensures that the channels on the Device and Host will coincide at some point. Further increasing the value has been observed to provide better performance in the presence of interferers.

So I believe we are both doing this wrong as `channel_table_size = 6` for receiver / Host. `6 channels in the table * 2 timeslots per channel * 900 or 600 us per timeslot * 6 channels (again! to cycle through them all)` is 64,800us or 43,200us to complete a full Host receive cycle when not in sync. This equates to 10,800us or 7200us on each of its six channels, remembering that only every second channel will match one Device. So this means that the window for a channel is effectively `10800us / 64800us = 17%` duty (although that may not be too bad). I will refer to such a period of reception on a channel when not in sync (before switching to the next channel) as a "sync-period" = e.g. the 10800us referred to above.

On the Device, in the meantime, it has tried transmitting `3 channels in the table * 2 timeslots per channel * 900 or 600us` = 5400us or 3600us for each of its three channels. This means if it starts on the channel after the matching one for the current "sync period" then by the time it gets round to the correct channel for the current "sync period", the "sync period" will have ended and it will be at least 10800us before another channel which matches this Device will be listened to again by the Host. 

So, I think the Nordic docs, with respect to `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*timeslots_per_channel)` is *only* referring to the Device but that the Host needs to be `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(1)` *unless* this parameter has *no* effect on the Host (which I cannot see mentioned anywhere so I assume it applies to  [... cut ...]
Finally (from this round of reading and thinking) I note that a larger channel table can help coexistence with interferers. I think I will consider this too for a future round of testing.

In summary, I will now try:

Device
- `nrf_gzll_set_timeslots_per_channel(4)`
- `nrf_gzll_set_timeslot_period(900)`
- `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*timeslots_per_channel*2)`
- `nrf_gzll_set_max_tx_attempts(20)`

Host
- `nrf_gzll_set_timeslots_per_channel(2)`
- `nrf_gzll_set_timeslot_period(900)`
- `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(1)`

I may implement these individually to identify which had most effect, if any.

karosc, 2020-12-15:
> So I believe we are both doing this wrong as channel_table_size = 6 for receiver / Host. 6 channels in the table * 2 timeslots per channel * 900 or 600 us per timeslot * 6 channels (again! to cycle through them all) is 64,800us or 43,200us to complete a full Host receive cycle when not in sync.

I am not understanding why we need to multiply by 6 twice. In my fork `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync(channel_table_size*(timeslots_per_channel+1))` was derived from components\properitary_rf\gzll\nrf_gzll_constants.h. The default value is 15 timeslots for a channel table of 5 and 2 timeslots per channel when in sync. Not sure why the default is larger than channel_table_size*timeslots_per_channel, but I upped mine accordingly.

SimonMerrett, 2020-12-15:
> I am not understanding why we need to multiply by 6 twice

We do not. I made a big error and have placed an EDIT at the top of my post above. In the Synchronisation section of the docs it is clear that the Host does not use `nrf_gzll_set_timeslots_per_channel_when_device_out_of_sync()`. Sorry for the confusion.

LucidityCrash, 2020-12-16:
@SimonMerrett My head just exploded :)

> Device
> 
>     * `nrf_gzll_set_timeslots_per_channel(4)`
> 
>     * `nrf_gzll_set_timeslot_period(900)`
> 
> 
> Host
> 
>     * `nrf_gzll_set_timeslots_per_channel(2)`
> 
>     * `nrf_gzll_set_timeslot_period(900)`

but isn't this where we started ... my pull request : https://github.com/mattdibi/redox-w-firmware/pull/7/files
note the device is 4 timestlots per channel with a 900 period and the host is 2 timeslots (the default so not explicitly set) per channel with a 900 period

SimonMerrett, 2020-12-16:
> but isn't this where we started

Yep, probably! Although going "back" to those settings is a huge improvement over what I had been putting up with for a few weeks, I still have the same odd missed / double-tapped keys. I briefly toyed with `nrf_gzll_set_sync_lifetime()` yesterday but along with a low number of retransmits, it seemed to lose a great deal of keypresses, so I turned it off and increased the retransmits from 20 to 100 (I had been using 200).

I think I will look at a larger channel table next, as that is specified as a fix for interference and we know that these setups are certainly vulnerable to other 2.4 GHz transmissions. I really should get some debug going on this keyboard, to know how many retransmits are attempted for example, how long the loops are taking (does the matrix scanning for the extra column/row on those of us with custom hardware cause race conditions etc?).

karosc, 2020-12-16:
I am still having the same problem. I need to start thinking about this a bit more technically I guess. It is very weird that my issue only occurs on the left half and only after a long idle period or after a pc restart. For some reason the left half falls out of sync with the host and cannot seem to get back into it.

SimonMerrett, 2020-12-16:
> It is very weird that my issue only occurs on the left half and only after a long idle period or after a pc restart

You could try swapping the order of the channels in the host list to see if a) different order within the channel table makes a difference and afterwards revert to the original order but change which channels left and right use to check b) whether the channels associated with the left are somehow less effective than the one's allocated to the right.

SimonMerrett, 2020-12-16:
I think there is some kind of debug through the qmk app/console but I haven't looked for a while.

LucidityCrash, 2020-12-16:
trouble is that is debugging at too high a level ... you really want to see what is going on in the nrf5 code
You can use openOCD and GDB together ... Its been a very long time but I seem to remember doing something that just put output in the OCD console.

karosc, 2020-12-16:
> Its been a very long time but I seem to remember doing something that just put output in the OCD console.

If you could provide some example code, that would be very helpful for me.

karosc, 2020-12-17:
So, I swapped the batteries on my keyboards and the issue started happening on the right half. So now this seems to have become a power delivery issue for me. I check the problem battery (a CR2032 coin cell) and found it was giving 2.89v, which is well within the 1.8 - 3.6v spec of the nRF51822. 

Now I am wondering if I need a second voltage regulator external to the nRF51822? Or maybe my batteries are just crappy quality, I just changed them last month.....they should not be dead.

danielo515, 2020-12-17:
> So, I swapped the batteries on my keyboards and the issue started happening on the right half. So now this seems to have become a power delivery issue for me. I check the problem battery (a CR2032 [...]

mmm, that is something very similar to what happens to me... not sure if my batteries are crappy, I just bought a bunch some time ago

[80 more replies]