mattdibi/redox-keyboard #56: three columns can not work

yoution, 2020-03-29:
I make a redox-wireless keyboard, the  right half  can work well, but in the left half, three columns can not work, others can work,  I don't know how to resolve the problem.

the keys  in the  red box , none of them can work
[image]

this is the back side
[image]

[image]

mattdibi, 2020-03-29:
Double check soldering.

Usually undetected keypresses on some columns are related to soldering problems. Judging from the YJ-14015 picture it seems a likely cause.

Hope this helps.

yoution, 2020-04-04:
@mattdibi  thanks，I screw up， the pcb panel was broken by my pincers， I need to remake the pcb， It will take one month maybe， I will try again after  I receive the pcb and  mcu

closed 2020-04-04 by yoution

yoution, 2020-04-18:
[image]

I am very excited it works, thanks you @mattdibi