mattdibi/redox-keyboard #60: Bricked my right keyboard after flashing

ddon, 2020-05-26:
followed instructions, was able to connected openOCD, was able to telnet to OpenOCD... was able to run "reset halt" and then "flash write_image"... then, after typing "reset" I didn't get any feedback, nor any errors... right keyboard stopped working.

i was doing this all on windows, and flashed only right keyboard using precompiled-basic-right.hex file from https://github.com/mattdibi/redox-keyboard/tree/master/redox-w/firmware

what did I miss? May be I need to flash left keybard as well? and may be even a receiver...?

reason I was flashing it, because I wanted to get fixed default english layout.

thank you for your help!

mattdibi, 2020-05-26:
Hi,

first of all to change the default layout **you should not flash the YJ-14015 microcontrollers**. The QMK firmware is responsible for the keyboard layout and resides on the Arduino Pro Micro receiver.

In general you don't need to use OpenOCD at all to use the keyboard or change its settings.

Secondly **you should not issue the OpenOCD commands directly** there are [two bash script for this](https://github.com/mattdibi/redox-w-firmware/tree/master/redox-w-keyboard-basic). Try running `program_right.sh` on the right half again.

ddon, 2020-05-26:
Hi,
Thanks for your reply here, still no luck, but had some progress...

Ok, I get it now, that I don't need to flash keyboard(s) in order to change the default layout, but since I did re-flashed it, how do I know if it is working or not working correctly?

Here are screenshots from flashing it, looks like everything worked correctly, please take a look, may be you will notice something strange:

[image]

[image]

Secondly, why I should not issue the OpenOCD commands directly? What is the difference if script sends them or I type them? (or copy paste them)  Reason I had to enter them manully, script it not working on windows ubuntu.

And lastly, you are offering to run program_right.sh, program_left and program.sh, but here I don't see program.sh script: https://github.com/mattdibi/redox-w-firmware/tree/master/redox-w-keyboard-basic

And for running program.sh, do I need to be connecting ST-LINK to the receiver?

ddon, 2020-05-26:
And do I need to download Nordic SDK and setup Toolchain if I just upload your .hex file to the keyboard and will not be doing any compiling?

ddon, 2020-05-26:
And from your other document, you wrote "You always need to burn the firmware on BOTH halves of the keyboard."... Could be the problem that I re-flashed only one keyboard? Can they conflict somehow? :)

mattdibi, 2020-05-26:
> why I should not issue the OpenOCD commands directly? What is the difference if script sends them or I type them?

I suggest to use the scripts because is less error prone.

>  you are offering to run program_right.sh, program_left and program.sh, but here I don't see program.sh script

The `program.sh` script which is used for the receiver is found [in the receiver source folder](https://github.com/mattdibi/redox-w-firmware/tree/master/redox-w-receiver-basic).

> And for running program.sh, do I need to be connecting ST-LINK to the receiver?

Yes, just like you did for the two Redox halves.

> And do I need to download Nordic SDK and setup Toolchain if I just upload your .hex file to the keyboard and will not be doing any compiling?

If you don't plan to recompile the code you don't need the toolchains.

> Could be the problem that I re-flashed only one keyboard? Can they conflict somehow?

If you flashed only one half my first hypothesis is that the original firmware has a different address than the one specified in the precompiled binaries. Your best bet is to flash the receiver and the other Redox half with the precompiled binaries.

ddon, 2020-05-26:
> If you flashed only one half my first hypothesis is that the original firmware has a different address than the one specified in the precompiled binaries. Your best bet is to flash the receiver and the other Redox half with the precompiled binaries.

Got it! Let me try... Only problem is that I need different cable to connect to a receiver... but I will find it! Will let you know how it goes...

ddon, 2020-06-04:
ok, good news! was able to put latest firmware and both keyboards now do work! thanks for your help... will now try to figure out how to fix US english layout...

ddon, 2020-06-04:
so, where is the tutorial how to change layout for this keyboard?

ddon, 2020-06-04:
So, since we didn't find any documentation on how to flash this particular keyboard with new layout, we went to official site QMK Toolbox project, there is download button, which downloads 20,000 files project, which takes 1 hour to unzip... and was complete waste of time, since this was not ready binary for windows, but looks like a source for the entire project.

Then we looked up for QMK Toolbox for windows, and found compiled .exe file, which was 10MB and we were able to run it.

We used QMK Toolbox online configurator, downloaded .hex file, and tried to flash redox keyboard with this .hex file using QMK Toolbox. Since there is no docs, we tried both ways, with received connected to ST-LINK and directly to USB, and in both cases, no luck.

Here is a screenshot of receiver connected via ST-LINK:
[image]

And here is a screenshot when receiver connected via USB:
[image]

So, what are we doing wrong here?

mattdibi, 2020-06-04:
Hi,
glad to hear that your keyboard is now working.

Regarding the layout: to update the QMK firmware on the Arduino Pro Micro **you should not use the ST-Link programmer**. That is for the YJ-14015 firmware only.

As I said the QMK firmware (which is the part responsible for reading the keyboard matrix state and sending the corresponding characters to the PC) is what you want to update and it is done **via USB**.

When ready to flash be sure to push the button on the Redox receiver which resets the Pro Micro and puts it **in the programming mode**. For everything else please refer to the [QMK documentation about uploading firmware](https://www.youtube.com/watch?v=VR53Wo9Z960).

mattdibi, 2020-06-04:
Glad I was able to help.

Enjoy your keyboard!

closed 2020-06-04 by mattdibi