mattdibi/redox-keyboard #67: test the Arduino

DickThomas, 2020-07-09:
Hello, 

I'm looking to build this keyboard for my birthday later on this month.
I keep reading the section regarding the 5v Arduino and to be honest it has me worried.
I'm not quite a total n00b to soldering, but I am at de soldering as it's something I've never done before.

Is there anyway I can test them preinstallation?

sorry for asking such a noobish question I know it's possibly not the right forum for this type of Q.

mattdibi, 2020-07-12:
Hi,

I think you're referring to the Pro Micro used in the Redox wired version. **Usually a simple visual inspection is sufficient to find out if the Pro Micro doesn't use the correct power cirtcuitry** (for example the wrong pro micro misses the J1 jumper).

[image]

More info [here](http://www.40percent.club/2017/09/green-pro-micro.html).

If you're referring to the 3.3V/5V, [you just need to change the CPU frequency parameter](https://github.com/qmk/qmk_firmware/tree/master/keyboards/lets_split#notes-on-using-pro-micro-33v).

DickThomas, 2020-07-16:
thank you so much
mine has a open J1 not a closed j1 as pictured, I was also worried about 3.3V/5V :+1: 

I've also found this supplier for this https://mechboards.co.uk/shop/components/24-pin-pro-micro-sockets/  that should solve the issue incase of a faulty board.

it's similar to the part you list but I was able to get it shipped to the UK for a reasonable price.

closed 2020-07-16 by DickThomas