mattdibi/redox-keyboard #68: one key not being detected

DickThomas, 2020-07-29:
Hello
I have made my redox keyboard and it is awesome.
but i have run in to one issue my " key on the right hand half  is not being detected. 

I have tested the switch and it is OK. 
the diode is the correct way around but seems  to be missing solder but I am not 100% sure. 

is it safe to put some more solder on the diode even though it is on the flip-side? 
are there any more diagnosing i can do?

many thanks in advance

closed 2020-07-29 by DickThomas

mattdibi, 2020-08-01:
Hi,

there's no issue in adding some more solder on a part if done properly (usual soldering rules apply).

Hope this helps.

DickThomas, 2020-08-01:
Hi,
I added it and all is working great now. Thank you so much for this project it has been an Awesome birthday gift to myself.
I've wanted a split keyboard for almost10years but didn't' dare commit until i saw your project on hackaday

thank you again