mattdibi/redox-keyboard #69: CNC case

AltusBaard, 2020-08-07:
Hi there,

So this is more of a request than anything else, I am busy building like my 4th redox, but I want to do something different, I want to have a case CNC'd out of aluminium, would it be possible to have a plate designed that will allow this?

The biggest issue that the CNC guys have are the little small recesses on the bottom side of the face plate for the switches to hook into. Unfortunately my CAD skills are not good enough to achieve this, would anyone be able to assist?

Regards

TheWeirdDodo, 2020-08-24:
Hi, just created this account to write.

I am working on a redox build myself right now and i made a few models. They are intended for 3D printing tho. If you are interested i can share them with you.

wsyxbcl, 2020-11-16:
> Hi, just created this account to write.
> 
> I am working on a redox build myself right now and i made a few models. They are intended for 3D printing tho. If you are interested i can share them with you.

If you'd like to make it open source, you can fork this repo and make your own commits. You can also submit a pull request based on that.

Pastitas, 2020-11-17:
I have a case and i'm planning to make it open source. based on the design i think it might easily be CNC cut if modded

LukeAndersonTrocme, 2022-07-13:
Reviving the thread, are the CNC compatible files available anywhere?

Pastitas, 2022-09-03:
I added my files to the [repo](https://github.com/Pastitas/Redox-neodox-Case), step files included, feel free to remix