mattdibi/redox-keyboard #70: Problem with CAD files

TheWeirdDodo, 2020-08-27:
Hi,

I'm right now in the process of building a redox keyboard and I wanted to model my own case and print it myself. It was very useful to have all the CAD files including the drawing of the PCB and the witch positions before hand to start modeling the case. 
So after designing a model I was happy with I printed it and put it together to see if it works out. Now after recieving my redox PCBs from falba I was very dissapointed when seeing that the holes for the switches on the top of the Keyboard don't align with the position on the PCB. Looking at my model and tryiung to find my error i found that the Error sits actualy within the dxf file found under the redox/case section. Sadly the dimensions of this drawing doesnt match the actual PCB from falba. 
I am really dissapointed by this after investing so much time into the design to find it isn't fitting. I dont know how to fix this problem, since i dont know the exact dimensions and offsets of the parts. I hope I can get some help here.

You can see what my problem is in the Image posted below. Marked in red you can see the position of the contacts where the switch will be soldered and in orange you can see the mounting position of the switches. Toward he left you can see that the position of both, the connections and the mounting holes align nicely centered, but toward the right you can see the position is shifted away from the center. I am 100% certain this error doesn't originate from bad 3D modelling but from wrong dimensions of the drawing and the actual PCB since i directly used the files from this repository to design this case.

[image]

I am hoping for a quick response, since this is changing my whole plan of building a nice keyboard using these PCBs.

mattdibi, 2020-08-29:
Hi,

sorry to hear you encountered these problems.

Unfortunately I haven't much to say about this. The method I recommend for designing new cases, and that worked until now for a lot of people, is to extract the position of the switches from the PCB in the .dxf format using KiCAD and starting from there with the design. Is this how you proceeded?

mattdibi, 2020-08-29:
Furthermore, did you try and install the switches (without soldering) to see if there's enough mechanical clearance for the case to work anyway? From the pic you posted the align doesn't seem to be as dramatic as you say. Might be a parallax error though.

TheWeirdDodo, 2020-09-03:
Hi,

sorry for the long wait. So i worked it out and it all fits, a little bit tight on the top right side of the board but it fits. I’m not completely sure what caused the problem. I just printed the same file again and now it works.

Thanks for the help tho!

closed 2020-09-04 by mattdibi