mattdibi/redox-keyboard #74: VIA compatibility

harshitgoel96, 2020-09-30:
Hi,

Would it be possible to make the QMK firmware for dongle be VIA compatible?

mattdibi, 2020-10-03:
Hi,

this is awesome!

>I can not do a PR because I had to change the VendorId and ProductId to make via work without clash.

I don't see a problem with it. When I wrote the firmware I left the default ones (which clashes with the GH60) because they weren't used by anything. Now with VIA things changed a bit. Feel free to modify it.

>Should I request a PR anyway ?

I think this is a really cool feature to add to the project. If you want to, please go on and request a PR to the QMK team.

harshitgoel96, 2020-10-03:
Ok. I will prep my branch for PR and let you know.
I already have a PR for ErgoTravel keyboard, will create Redox-w PR once ErgoTravel is pulled into qmk master

harshitgoel96, 2020-10-03:
Incase you want to try out via let me know will share the config files with you.

harshitgoel96, 2020-10-03:
Hi @mattdibi  

I have added config to my branch of QMK and VIA. Please use the following to test. it will be a while before qmk pulls the changes for ErgoTravel. Only after that I can push my changes for Redox_w

QMK:
https://github.com/harshitgoel96/qmk_firmware/tree/redox-via

VIA Json:
https://github.com/harshitgoel96/keyboards/tree/redox_w-via

I will update the thread once PR has been raised.

harshitgoel96, 2020-10-08:
@mattdibi 
I have created pull request for redox_w via.

https://github.com/qmk/qmk_firmware/pull/10578

harshitgoel96, 2020-10-17:
@mattdibi the via config is merged in QMK master.

TheWeirdDodo, 2020-10-20:
Is it possible to get via support for the non wireless redox? I looked into it but the via documentation is kinda poor and I don’t fully understand what to do.

harshitgoel96, 2020-10-21:
> Is it possible to get via support for the non wireless redox? I looked into it but the via documentation is kinda poor and I don’t fully understand what to do.

Yes it should be possible. What microcontroller does it use?

TheWeirdDodo, 2020-10-21:
> > Is it possible to get via support for the non wireless redox? I looked into it but the via documentation is kinda poor and I don’t fully understand what to do.
> 
> Yes ~it should be~ is possible. ~What microcontroller does it use?~ it uses pro micro, I will prepare a config and put on my repo. Since I don't have the hardware someone will help me in testing it.

Ok cool, I’m also using an elite-c in one of my halves, but this controller is a drop in replacement for the pro micro. I can test it on my keyboard then when you got the config ready, I'd just need some minor instructions on what to do with the config file.

harshitgoel96, 2020-10-21:
@TheWeirdDodo 
Pro Micro and Elite C are based on same ATMEGA32 microcontroller so its not an issue.

I have done config on 

https://github.com/harshitgoel96/qmk_firmware/tree/redox-harshit

* please compile using `make redox:via`
* flash normally.
* once flashed open via, go to `settings-> enable design lab-> switch to design tab -> click on load-> select via.json from redox folder`

if everything was correct you should not be able to enjoy VIA features on redox.

Please let me know if any issue is present.

Pastitas, 2020-10-21:
This might not work as both halves of the keyboard have intependent pro micros, but i might be wrong

TheWeirdDodo, 2020-10-21:
@harshitgoel96
Thanks, I will give it a try!

TheWeirdDodo, 2020-10-21:
I am back with good news, it works just fine. Was able to flash it with my firmware and via support! thanks a lot for your help!

harshitgoel96, 2020-10-21:
> I am back with good news, it works just fine. Was able to flash it with my firmware and via support! thanks a lot for your help!
> Only "problem" is that via reports it as a redox-w but this is only in the editor.

I have updated the via.json with correct name. Please take an updated from the branch. Please confirm all keybinding are correct so that I can proceed with pull request on master qmk branch.

TheWeirdDodo, 2020-10-21:
> I have updated the via.json with correct name. Please take an updated from the branch. Please confirm all keybinding are correct so that I can proceed with pull request on master qmk branch.

Thanks for the update. All keybindings worked fine. All keys are recognized correctly in via.

harshitgoel96, 2020-10-21:
> Thanks for the update. All keybindings worked fine. All keys are recognized correctly in via.

@TheWeirdDodo  thanks for testing. i have requested a PR
https://github.com/qmk/qmk_firmware/pull/10710

harshitgoel96, 2020-10-21:
@mattdibi  once this pull is merged with qmk i will close this issue.

TheWeirdDodo, 2020-10-22:
@harshitgoel96
Hi, I just heard from a friend that he can controll his keyboards rgb light using via, I also got rgb for my redox but there is no option in via for this. Do yo know what i need to do to implement this?

harshitgoel96, 2020-10-22:
@TheWeirdDodo  RGB support is enabled on VIA.

I think special keys you should find RGB keys, and you should be able to map it to layers.

TheWeirdDodo, 2020-10-22:
> Can you ask your friend which keyboard he is getting option to change lighting from via ?

So it is a Switch couture alice with a pcb from Project keyboard. According to the via docs, a setting in the via.json is needed:
`"lighting": "none",`
Here then one would need to set: `none,` `qmk_backlight,` `qmk_rgblight,` `qmk_backlight_rgblight,` `wt_rgb_backlight,` or `wt_mono_backlight` depending on what type of lighting is used.

TheWeirdDodo, 2020-10-22:
Ok so I got it working now in via, had to change `none` to `qmk_rgblight` in the via.json file.

harshitgoel96, 2020-10-30:
@TheWeirdDodo  @mattdibi  the PR has been merged for OG Redox. i will close this ISSUE now

closed 2020-10-30 by harshitgoel96