mattdibi/redox-keyboard #77: Arduino turns off when TRRS cable is connected

nicola-sorace, 2020-12-12:
Hello

I've assembled a non-rgb keyboard according to the instructions, and flashed the firmware on each half. Each individual keyboard half works perfectly on its own, but when I connect them with the TRRS cable the Arduino LEDs turn off and nothing happens when I type. This happens even if I plug in a cable into only one side and **leave the other end unconnected**.

I've tried a different cable, double-checked all connections are as they should be, but nothing seems to work. If I pull out one side of the cable ever so slightly, the LEDs on both Arduinos turn on, but only the side that's connected actually works. If I use two separate micro USB cables to connect both sides independently, everything works great.

Any ideas what has gone wrong?

mattdibi, 2020-12-12:
Hi,
this sounds like a short to ground or a faulty connection caused by the TRRS connector. Double check with a multimeter in continuity mode that the TRRS connector does what it should and there's no shorts in its pins.

nicola-sorace, 2020-12-12:
Hi, thank you very much for the fast reply!

I've checked everything I can think of with a multimeter. The TRRS connector seems to be wired up like this:

[image]

If that looks right then it seems like everything is wired up like in the PCB schematic (I'm no expert though). Only weird thing is that the resistance across the resistors is about 4.56kOhms instead of 4.7kOhms? Other than that everything seems in order. When I connect the cable the matching TRRS pins on both keyboard halves are fully connected.

Just tried connecting the TRRS cable before plugging in the micro USB. No keys do anything. As soon as I unplug the TRRS the half that's plugged in works fine.

nicola-sorace, 2020-12-12:
ok so, it looks like when a jack is plugged into a socket, the pins I labelled "ground" and "VCC" in the image are shorted. I tested this on a loose socket and the same thing happened.

Did I buy the wrong type of connector? It's this one here: https://www.amazon.co.uk/sourcing-Connector-Female-Socket-PJ-320A-Black/dp/B07KYCC7RN/ref=cm_cr_arp_d_pl_foot_top?ie=UTF8

Surely the circuit board can't be wrong, and I don't see how I could have soldered in a way that switches pins around (and in the same way on both sides)! What's going on?

nicola-sorace, 2020-12-13:
...so it turns out I was using a T**R**S cable instead of a T**RR**S cable. The other cable I have *is* a TRRS cable, but for some reason *one* of its sides has this same effect of shorting out VCC and ground.

So it looks like my problem might have been *two* faulty cables! 🤦
I'll order a new one and see if that solves the problem

mattdibi, 2020-12-13:
My next move would have been to suggest you to check the cable, nice catch!

Closing this. Feel free to re-open this if changing the cable doesn't solve the problem.

closed 2020-12-13 by mattdibi

nicola-sorace, 2020-12-15:
Confirming that was the problem, new cable works great.
Thanks for the help! This keyboard is awesome!