mattdibi/redox-keyboard #80: Question about RGB and i2c

renkam, 2021-01-27:
If I understand everything correctly when LED underglow is not used, then both sides communicate over i2c. If underglow is used then serial connection is used instead i2c.

On schematic pdf (https://github.com/mattdibi/redox-keyboard/blob/master/redox/pcb/Redox-schematic.pdf) there is rgb_data output. Is this output uses i2c? Is it possible to use oled display with rgb underglow in this configuration?

mattdibi, 2021-01-30:
Hi,

A clarification. QMK [allows for two type](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_split_keyboard) of communication for split keyboards: serial and I2C. I2C needs **two** data wires (SDA, SCL) to work while serial only **one**. To drive the LEDs we need that extra wire and thus if you use the I2C communication you can't connect the LED underglow.

The `rgb_data` output is a single data output and thus cannot be configured to drive I2C-based OLED displays (because again they need 2 data lines).

Hope this helps.

renkam, 2021-01-31:
This helps.
If I understand correctly (again :smile:), I can [use RX1/PD2 for serial](https://user-images.githubusercontent.com/2170248/92296490-2d15cb00-ef70-11ea-801f-5ace313013e6.JPG) and then I have 2/PD1 and 3/PD0 free to use for OLED display. Also I have still free 5/PC6 and A3/PF4 for rotary encoder.

Please correct me if I am wrong.

mattdibi, 2021-01-31:
Technically yes. But you would need to modify the Redox PCBs as these pins are already used (PD1 and PD0 are routed to the TRRS connector).

renkam, 2021-02-02:
Understood. Thanks!

closed 2021-02-02 by renkam