mattdibi/redox-keyboard #83: weird random key combinations

MrPelucaa, 2021-02-18:
So just finished the wirign, everything is ok but when flashing my board, in seems in sends random combinations of keyes. After re checking everything now it seems it has some keys stuck. Im pretty sure the problem is in the flashing process but I dont really understand it.

Platform: windows
Board: Pro Micro Clone

 Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
        Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.
    
    Programmer supports the following devices:
        Device code: 0x44
    
    avrdude.exe: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.00s
    
    avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
    avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
                 To disable this feature, specify the -D option.
    avrdude.exe: erasing chip
    avrdude.exe: reading input file "C:\Users\diego\Downloads\redox_rev1_default_4ac56f60.hex"
    avrdude.exe: writing flash (24400 bytes):
    
    Writing | ################################################## | 100% 1.80s
    
    avrdude.exe: 24400 bytes of flash written
    avrdude.exe: verifying flash memory against C:\Users\diego\Downloads\redox_rev1_default_4ac56f60.hex:
    avrdude.exe: load data flash data from input file C:\Users\diego\Downloads\redox_rev1_default_4ac56f60.hex:
    avrdude.exe: input file C:\Users\diego\Downloads\redox_rev1_default_4ac56f60.hex contains 24400 bytes
    avrdude.exe: reading on-chip flash data:
    
    Reading | ################################################## | 100% 0.17s
    
    avrdude.exe: verifying ...
    avrdude.exe: 24400 bytes of flash verified
    
    avrdude.exe done.  Thank you.

mattdibi, 2021-02-19:
Hi,

from experience I can tell you that these kind of problem almost always come from some hardware issues. Double check soldering for shorts and whatnot. Double check diodes orientation.

The stuck keys are all on the same column? All on the same row? That can be a hint on where the problem can lie.

I don't see any problem in the firmware upload process.

MrPelucaa, 2021-02-19:
After checking the connections it seems the arduino has continuity (?) between the pins a2 and 3. Also im not sure it means continuity, the multimeter shows numbers over 1000 and no beep, like when testing a diode. That has to be the problem, because when I tested in the keyboard that means a full column is constantly being pressed. Not sure if the arduino is faulty or my soldering. Is there a way to remap the pins in QMK ?

MrPelucaa, 2021-02-19:
just switched to another arduino, works fine. Guess it was bad luck with that board.

closed 2021-02-19 by MrPelucaa