mattdibi/redox-keyboard #86: Latest revision of PCB from Falba

smazur1974, 2021-02-21:
So I got latest revision from falbatech and I have to say that line 5V and GND (LED) are reverted but there is square under 5V... After swapping cables everything is working - there should be info regarding this, here in this guide

mattdibi, 2021-02-21:
Hi,

this comes as news for me. I’m not aware of a new version of the Redox keyboard. Would you mind sharing some pictures of the board? I’m particularly interested on he differences you mentioned...

smazur1974, 2021-02-21:
Well, I it is too late cuz I have already build it. There are two changes. One intended - the diodes which were between Pro Micro are now moved next to it. The second thing is that as I said  the LED lines are reverted (5V and GND). There is mistake (I think) cuz 5 V line is marked as rectangle. And I followed steps from your guide and I was suprised that keyboard was dead. After some tinkering and desoldering LED cables I was able to make it working and from there it was pretty easy.

smazur1974, 2021-02-22:
[image]
[image]

mattdibi, 2021-02-22:
Thank you very much for posting this pictures. I didn't know Falbatech added hot swap support! I'll try and add a note about the LED lines in the docs ASAP.

Spamm3r, 2021-02-22:
This my private account and from which I shall continue this ticket. I have found one more difference - Falba has inverted keys which are "above" pro micro. They are now facing sideways (You can clearly see that pads which are under pro are in different position) but I think that there is no much difference outside of that
[image]
.

Pastitas, 2021-02-23:
>I have found one more difference - Falba has inverted keys which are "above" pro micro. They are now facing sideways (You can clearly see that pads which are under pro are in different position) but I think that there is no much difference outside of that.

I supose this is done so the hotswap sockets fit there, so yeah ,shouldnt make any difference. That's probably also why they've moved the diodes outside.

@mattdibi I am not sure under wich license you published this PCB  design, but shouldn't falbatech share this updates with the community? I myself think that they should, but I don't know what kind of commercial agreement (If any at all) you guys came to, so I might be wrong.

mattdibi, 2021-02-23:
@Pastitas 

> I am not sure under wich license you published this PCB design, but shouldn't falbatech share this updates with the community? I myself think that they should, but I don't know what kind of commercial agreement (If any at all) you guys came to, so I might be wrong.

Everything regarding the Redox is MIT licensed so he can do basically what he wants to. If he doesn't want to share his design and profit off of them he can.