mattdibi/redox-keyboard #88: Falba PCB revision manufacture date 13-01-2021

raymaclabs, 2021-02-23:
I recently received my boards from Falbatech and noticed an issue with an overlap of a hotswap pad, a diode smd pad and the associated diode through hole.  It looks like it was scratched through the copper layer on the "left" side to try and correct the issue.  Has anyone that has built with this board had any issues?  I am doing through hole and not hot swap, so I am not overly concerned but wanted to validate.  Looking at the other post here with this board revision I can see the same scratch mark.  Attached are photos of both boards with the same scratch on the "left" as well as the flip side of the board with properly spaced pads on the "right" side of the board.

Thanks for a great design and offering the PCBs at a reasonable cost compared to a lot of other kits.

[image]
[image]

mattdibi, 2021-02-23:
Hi,

with the premise that, as emerged from [here](https://github.com/mattdibi/redox-keyboard/issues/86), I've just learned about the existence of this particular board. It looks like there was a misfortunate placement of the pads that was later fixed by cutting the trace after the fact.

I think that, if you're not planning to use the hot swap socket, you should be good but I cannot be 100% sure.

To be extra sure you can grab the multimeter and make a few continuity checks using the schematic available [here](https://github.com/mattdibi/redox-keyboard/blob/master/redox/pcb/Redox-schematic.pdf).

Spamm3r, 2021-02-23:
Okay, about this "scratches", I have exactly the same in same exact places but board is functioning properly. I have even checked for continuity and everything was alright. If you are planning installing LED's then you should read my post.

raymaclabs, 2021-02-23:
Thanks @mattdibi and @Spamm3r for the comments.  I haven't done the continuity check but will do that tonight and validate.  Good to hear its working fine for you Spamm3r.

Matt since I haven't done the install yet do you want me to take photos of the build process so you can add it to your documentation with this updated board?

Spamm3r, 2021-02-23:
I think that there is no need to update everything. the most important parts are that hot swap mod is supported and LEDs lanes are inverted. But decision is up to Matt.

@raymaclabs how long have you wait for your order?

raymaclabs, 2021-02-24:
@Spamm3r it didn't seem unusually long.  It was 4 days from order to shipping and then less than a week to arrive in the USA.

Spamm3r, 2021-02-24:
Oh really? I was waiting month or even more. I have ordered case, cables and pcb.

Spamm3r, 2021-02-25:
@mattdibi I may have problem with one key (pad is overlapped) (key is ]/} - I have default keymap). I have checked for continuity when I have pressed key and there is connection between pads. I have replaced diode and different positions but no success (you never know if falba changed something). So is there an way to simulate an key press by shorting some pads on pro micro? If traces are damaged I will just short wire connection to key.

mattdibi, 2021-02-25:
Great! Thank for the informations. I'll wait for the documentation from Falbatech then.

closed 2021-02-25 by mattdibi

Spamm3r, 2021-02-25:
I have no clue how long you will have to wait etc. Maybe if you will call him you will get more answers then me.

raymaclabs, 2021-02-25:
@Spamm3r is this the spot with the scratched pads or a different one?  I want to ensure i jump the same spot when soldering mine up also.  Thanks!

Spamm3r, 2021-02-25:
Yea with scratched pads. The funny thing is that on my one half everything is alright but on second one is not. I can guess that he did "disconnecting" of pads by hand and in my case he damaged trace.

raymaclabs, 2021-02-25:
I see what you're talking about, the trace for that last column that is close to those pads.

Pastitas, 2021-02-26:
Did you guys buy these as B-Stock units for a lower price, or are they the standard option?

raymaclabs, 2021-02-26:
I bought these as the standard option from this link. https://falba.tech/product/ergodox-pcb-electrical-boards-set-of-2/

mattdibi, 2021-02-27:
@raymaclabs But... those are Ergodox PCBs not Redox PCBs.

Spamm3r, 2021-02-27:
I have bought mine as normal PCB's. So dunno.

raymaclabs, 2021-03-01:
Sorry bad cut and paste on my end. It was the redox ones here. There is no
listing for a b-stock pcb of redox.

https://falba.tech/product/redox-pcb-electrical-boards-set-of-2/