mattdibi/redox-keyboard #91: Right hand cable

Vicepredator11, 2021-03-18:
Hi,

i've build my redox keyboard, but i would like to connect the keyboard on the right side, with the precompiled firmware i find myself with a reverse keyboard, can you please help me?

Thanks
Erik Vicenzotto

mattdibi, 2021-03-19:
Hi,

please refer to [this document](https://github.com/mattdibi/redox-keyboard/tree/master/redox#setting-ee_hands-to-use-either-hands-as-master) from the Redox documentation (which seems to be a little outdated). Also refer to the [QMK documentation](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_split_keyboard#handedness-by-eeprom) for latest QMK version.

Vicepredator11, 2021-03-19:
I made it possible by only add this line
```c
#define MASTER_RIGHT
```

thx for your help

closed 2021-03-19 by Vicepredator11