mattdibi/redox-keyboard #92: Defining custom layout

ftassi, 2021-03-22:
First of all thanks for the great work, I've built my first redox and I'm really enjoying it right now. I'm struggling a little bit figuring out the best way to customize my keymap though (maybe I'm just a qmk newbie). 

What is the best tool for the job? 
I am playing with https://config.qmk.fm/#/redox_w/LAYOUT which seems really a great tool, but I can't figure out how to compile right and left hex files. The configurator seems to be generating a firmware for the whole keyboard. Shall I do some manual manipulation to the keymap.json file before compiling it? Is there a better graphical tool? Should I not use a graphical tool at all ?

mattdibi, 2021-03-22:
Hi @ftassi,

the keyboard uses a single firmware version for the two halves, it auto-detects which is the plugged one and uses that as master (if you're using a wired one). Regarding the graphical tool you can also use [VIA](https://caniusevia.com/) (be warned that you need to install [the correct firmware ](https://caniusevia.com/docs/download_firmware#0) first to leverage the tool).

Hope this helps.

Saluti,
Mattia.

ftassi, 2021-03-22:
> the keyboard uses a single firmware version for the two halves, it auto-detects which is the plugged one and uses that as master (if you're using a wired one)

Actually I'm using the wireless redox, and I flashed two different hex files for left and right transmitters, following your instructions. Am I not supposed to generate a new left and a new right file to change the keymap?

_BTW I'll look into via as well, I don't know it_

mattdibi, 2021-03-22:
> Actually I'm using the wireless redox, and I flashed two different hex files for left and right transmitters, following your instructions. Am I not supposed to generate a new left and a new right file to change the keymap? 

Note that you're talking about two different firmware here: there's the YJ-14015 firmware and there's the QMK firmware (which is used by the ProMicro). To change layout you only need one .hex which is generated by QMK via VIA (pun not intended).

[Here](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w#firmware) some more details about the firmware and the MCUs of the Redox Wireless.

ftassi, 2021-03-22:
Oh silly me :facepalm:  you're right. I just need to update the ProMicro to flash a new keymap. Thanks

mattdibi, 2021-03-22:
Feel free to re-open this if you have any more doubts. :)

closed 2021-03-22 by mattdibi