mattdibi/redox-keyboard #93: Redox W Issues

Goldielocks, 2021-03-25:
I'm having an issue with the receiver: I can program it by header, but when I connect it to the Arduino Pro Micro, no lights appear. I've assembled several units at this point, since I ordered some redundant parts, and they all have the same behavior. If pin  validated the continuity (testing for shorts as well) but the lights still do not turn on.

I record a voltage drop over the LED/resistor as well, so it seems odd that nothing is visible. Do you have any thoughts?

Goldielocks, 2021-03-25:
https://photos.app.goo.gl/ZBaQ3kaNugr6mHov6
https://photos.app.goo.gl/jb4UTSCqM7cQC6G86

I've confirmed with Falba that their PCB's are behaving correctly. The above pictures are of one of the assembled boards.

Goldielocks, 2021-03-25:
Testing the voltage on my pins, it's becoming clear that none of them are reaching a sufficient low for current to pass. They seem to be floating. It's possible that the arduinos are no good. At least one of the ones I was testing with turned out to be the 3.3V Pro Micro

mattdibi, 2021-03-25:
> It's possible that the arduinos are no good. At least one of the ones I was testing with turned out to be the 3.3V Pro Micro

Yup. You should use 5V Pro Micros.

Goldielocks, 2021-03-25:
Sorry, this is not closed. The 5V Pro Micro has not solved my issue

mattdibi, 2021-03-26:
From #94 

>  I think I've confirmed that the problem is on the Arduino/QMK side of things. It's an arduino micro clone, but it's still using an Mega32U4-MU. Arduino IDE identifies it as a leondardo, which I understand is fairly common for micro clones.

Did you pinpoint the issue? What resistor values did you use for the LEDs? Does the receiver works (i.e. keystrokes are registered?) Does this happen for all the LEDs?

Goldielocks, 2021-03-26:
The issue hasn't been pinpointed, but at least one receiver is working. Only the first one appears to have been bricked, which led me down a bit of a goose chase. In fact I'm writing on it now. 

I'm referring to the LEDs on the receiver board. Now that it's transmitting, it appears that the bottom LEDs do function. I'm using the default configuration for the LED/resistor pairs. Red, Blue, Green, White in that order top to bottom. 330, 220, 330, 220.

mattdibi, 2021-03-27:
> I'm referring to the LEDs on the receiver board. Now that it's transmitting, it appears that the bottom LEDs do function. I'm using the default configuration for the LED/resistor pairs. Red, Blue, Green, White in that order top to bottom. 330, 220, 330, 220.

The LEDs light up when the layer is switched. To trigger a layer switch you need to press the following keys in the default keymap:

[image]

Refer to [the default keymap ](https://github.com/qmk/qmk_firmware/blob/master/keyboards/redox_w/keymaps/default/keymap.c)for details.

Please note that the fourth LEDs is not mapped to any function in the default keymap and thus won't light up.

>  I'm using the default configuration for the LED/resistor pairs. Red, Blue, Green, White in that order top to bottom. 330, 220, 330, 220.

You should be good then.

TheCatster, 2021-04-05:
I am having similar issues, I am using the SparkFun Pro Micro. However, I do not have any LEDs lighting even after following this issue. I have checked the leads and all seems proper, and my reset button works, but the LEDs do nothing. Any ideas for that (or should I make a new issue)?

mattdibi, 2021-04-05:
@TheCatster At this point I’m suspecting something broke in QMK. If you burn the pre compiled QMK hex files available [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w/firmware) do the LEDs light up? That file (`qmk_redox_w_default.hex`) was built using an older version of QMK and should work.

TheCatster, 2021-04-05:
I'll try it out as soon as I'm done flashing the left hand here. To add to your possible suspicion, I haven't actually tried your QMK file yet, I have only compiled my own keymaps!

mattdibi, 2021-04-06:
Just tested it with the latest version of QMK firmware. Everything is working fine.

@TheCatster If you can't connect the transmitters to the receiver the LEDs won't light up as they're triggered by key presses. You won't get keypresses until the firmware for the trasmitters MCUs, the firmware for the receiver MCU and the QMK firmware for the Arduino are all burned on their MCUs as described [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w#firmware). If you still have issues, please open a new issue.

@Goldielocks Since I didn't receive any update I'm closing this. Feel free to reopen this if you still have issues.

closed 2021-04-06 by mattdibi

tterrag1098, 2022-03-20:
Hi, I'm currently running into this issue. I can only get the LEDs to work if I use the default firmware from the redox repository. If I try to compile it myself (using 0.15.25 due to https://github.com/qmk/qmk_firmware/issues/16553) everything seems to work fine *except* the LEDs on the arduino. Is there anything in QMK that could have changed to break this? Or a setting that needs to be changed from default?

tterrag1098, 2022-03-20:
Just to sanity check, I tried with the version of QMK that would have been latest at the time of your last comment, 0.12.32, and found the same behavior.

tterrag1098, 2022-03-20:
Okay, sorry for the triple post here, but I wanted to update you to say I found the solution.

The root cause is definitely that I am a noob, but I think there is a serious lack information here. It took me many hours to find that the LED behavior is implemented [in the keymap](https://github.com/qmk/qmk_firmware/blob/master/keyboards/redox_w/keymaps/default/keymap.c#L91-L109) at the very bottom. The very limited information about this I can find online never mentions this, and in fact seems to imply that it's built into the firmware.

All the guides to first time users creating their first firmware suggests to use the online configurator, which of course creates a keymap file from scratch, and won't have the above code in it. Is there no way this could be implemented into the keyboard-specific firmware code in QMK, rather than the keymap?

Again, sorry for all the comments about this, but I wanted to get all this info out there for others in the future, and in the hope it might be improved. Thank you for all your work on this keyboard, it really is great to use. This issue is minor in the grand scheme, after all.

TheCatster, 2022-03-21:
That's quite interesting. I actually gave up on the LED, thinking it was something I had done wrong (I guess in a way it was). This would be nice to get in the firmware, but as @tterrag1098 said it's a minor issue overall.