mattdibi/redox-keyboard #94: Redox W Receiver Shows No Status Lights

Goldielocks, 2021-03-25:
Copied from the original issue:

I'm having an issue with the receiver: I can program it by header, but when I connect it to the Arduino Pro Micro, no lights appear. I've assembled several units at this point, since I ordered some redundant parts, and they all have the same behavior. If pin validated the continuity (testing for shorts as well) but the lights still do not turn on.

I record a voltage drop over the LED/resistor as well, so it seems odd that nothing is visible. Do you have any thoughts?

https://photos.app.goo.gl/ZBaQ3kaNugr6mHov6
https://photos.app.goo.gl/jb4UTSCqM7cQC6G86

I've confirmed with Falba that their PCB's are behaving correctly. The above pictures are of one of the assembled boards.

Goldielocks commented 3 hours ago
Testing the voltage on my pins, it's becoming clear that none of them are reaching a sufficient low for current to pass. They seem to be floating. It's possible that the arduinos are no good. At least one of the ones I was testing with turned out to be the 3.3V Pro Micro. Using a 5V micro does not solve this problem, but I notice that all 4 pins are sitting fairly high. If I ground any one of them, the respective light turns on.

Goldielocks, 2021-03-25:
I think I've confirmed that the problem is on the Arduino/QMK side of things. It's an arduino micro clone, but it's still using an Mega32U4-MU. Arduino IDE identifies it as a leondardo, which I understand is fairly common for micro clones.

mattdibi, 2021-03-26:
Closed as duplicate of #93 . Please continue on the original issue.

FYI you should have the permission to re-open an issue if it isn't solved.

closed 2021-03-26 by mattdibi