mattdibi/redox-keyboard #95: One column in the left side of the keyboard is malfunctioning

cyanindya, 2021-03-28:
Hello, apologies for opening another issue, but I'd like to hear your thoughts before going through the drastic measure of replacing the Pro Micro, since I'm not quite sure yet whether the fault lies in the Pro Micro itself or soldering on the PCB board.

So since this morning, one column in the left half of my Redox (5, t, g, b, thumb cluster button assigned to "left shift on hold and space on tap") started malfunctioning. Specifically, it suddenly became dead, and when I tried to re-plug the keyboard, the keyboard ended up inputting the whole column when I tried to press random key outside of that column (e.g. "a"), sometimes continuously until I somehow managed to stop it. The column stays dead, and it also affected another key in the other side (the outer-most 1.5u thumb key in the right half - I assigned it to "backspace", but it ended up inputting left shift as if the key was being hold down).

So far, for the hardware check, I already tried checking the diodes on the specified column and the pins connected to the keys in said column. I hadn't encountered problems possibly caused by improper soldering and such on the PCB so far - all diodes show the proper measurement when I tried using multimeter, and the row and column pins' connection seem to behave as expected (i.e. in that the targeted row and column pins are connected when I pressed the key switch down, but not the others). Because of this, I suspect the problem lies in the Pro Micro of the left half.

The thing is... I also tried crude method to figure out the problem, such as using jumper to directly short the targeted row and column pins, and doing the connectivity test on said pins using multimeter while the keyboard is plugged to the PC. For whatever reason, after trying this, the dead column problem will suddenly fix itself temporarily. The keyboard will work normally again for some time (10-20 minutes) before the dead column suddenly occurred again, again preceded by character input from the whole column even though I wasn't pressing any keys on that column.

What are your thoughts on this, and is it possible I may have missed something?

Thank you!

mattdibi, 2021-03-28:
Hi,

the problem you describe looks awfully like a cold joint.

These are particularly annoying to deal with because are hard to detect. In the past I had to deal with a cold joint on my job, it went unnoticed until some component stopped working all of a sudden (it was on an I2C connection). Poking around with a multimeter didn't show anything because the act of pushing the probes on the joint closed the circuit thus solving the problem. We found out because the I2C connection came back up when pushing on the PCB with a finger.

I suggest you to double check the solder joint of the affected column and the related pins on the Pro Micro. Look more closely on the Pro Micro side of things.

Hope this helps.

cyanindya, 2021-03-28:
Thank you for your input! It didn't occur to me that the cold joint may be the root of the problem.

As of now, I fixed the problem by replacing the Pro Micro with a new one since the unprompted input of the column persisted even when I took the switches of the column off and reflowing the Pro Micro pins didn't seem to have any effect (hence I suspected something might have gone off inside it), but I shall keep close eye on the solder joints of the affected parts as you said just in case.

Thank you again! I will close the issue for now.

closed 2021-03-28 by cyanindya