mattdibi/redox-keyboard #96: Keyboard is not responsive until I turn it off and on again

ftassi, 2021-03-29:
Hi I am experiencing a strange behaviour with my redox: after some usage the keyboard seems to be less responsive than the usual, key pressed are sent with some delay or not at all, and sometimes multiple key are sent all at once. Usually if I turn of and on the affected half it get back on track for a while.  I'm getting to the point were I restart the half twice a day. I don't think that the keyboard is "getting worse", but rather than I am noticing the issue more often because I am using it more and more. I noticed issue #55 where the described issue seems similar to mine, but I don't get how some interferences could be solved by restarting the keyboard.

Is there anything I can do to debug the issue? Is it a known behaviour?

ftassi, 2021-03-30:
Yes I am using bamboo case and plate from falba(plate should be steel) . And yes I was wearing wifi headset (I always keey headset receiver on a separated USB hub) 
Batteries are brand new, I assembled the keyboard a week ago (I could try to sobstitute them anyway). I noticed that the left half is failing more often than the right one, and I'm keeping the receiver near the right one (we're speaking centimeters here), so could indeed be an interference issue :(

I have repositioned the receiver nearer the TX and right in the middle of the two halves. We'll see it this change something

mattdibi, 2021-03-30:
> Batteries are brand new, I assembled the keyboard a week ago (I could try to sobstitute them anyway).

I wouldn't expect them to make a difference if they're that new. From experience some discount brands perform worse but yeah, it shouldn't make so much difference.

> Yes I am using bamboo case and plate from falba(plate should be steel) 

That's what I was thinking. I recently switched from Falbatech's bamboo and steel case to a 3d printed one and the reception is much better. Unfortunately the case can play a role in this.

> I have repositioned the receiver nearer the TX and right in the middle of the two halves. We'll see it this change something

Let me know if this improves your reception.

ftassi, 2021-04-02:
having the receiver right at the center of the keyboards seems to help. I have not experienced any other freeze with this setup. I am not sure this solves the issue entirely though because I am not nearly as fast as I should be with the redox yet. Anyway I guess this could be enough to close this issue for now. I'll get back to you when I'll get a decent wpm to test the communication issues. In the meantime I could write an email to falba, just the let him know the issue, maybe he could figure something out. I am wandering if a plastic case for the receiver only  could help here. I would really stick to the bamboo case for the halves, I really love them ;)

closed 2021-04-02 by ftassi